FastLED/FastLED #192: Compile errors w/ Arduino 1.6.5 on AVR tiny167

Zitt, 2015-07-21:
Trying to port a FASTLED project from AVRtiny85 to tiny167 as I needed more flash to compile both FASTLED and SerialUSB (DigiCDC) in a single project. Getting some strange error messages which I wasn't expecting.

Arduino: 1.6.5 (Windows 7), Board: "Digispark Pro (Default 16 Mhz)"

#define RORSC24(B, N) ROR1(B) CLC1 SCALE22(B, N)
                               ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:297:27: note: in expansion of macro 'RORSC24'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                           ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:160:103: error: impossible constraint in 'asm'
 #define SCALE22(B, N) asm __volatile__("sbrc %[s2], " #N "\n\tadd %[" #B "], %[scale_base]" ASM_VARS );
                                                                                                       ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:183:36: note: in expansion of macro 'SCALE22'
 #define RORSC24(B, N) ROR1(B) CLC1 SCALE22(B, N)
                                    ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:297:27: note: in expansion of macro 'RORSC24'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                           ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:150:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:297:48: note: in expansion of macro 'LO1'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                                                ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:175:60: error: impossible constraint in 'asm'
 #define ROR1(B) asm __volatile__("ror %[" #B "]" ASM_VARS );
                                                            ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:297:52: note: in expansion of macro 'ROR1'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                                                    ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:177:47: error: impossible constraint in 'asm'
 #define CLC1 asm __volatile__("clc" ASM_VARS );
                                               ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:297:61: note: in expansion of macro 'CLC1'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                                                             ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:148:110: error: impossible constraint in 'asm'
 #define HI1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[hi]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=hi; }
                                                                                                              ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:298:5: note: in expansion of macro 'HI1'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
     ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:152:69: error: impossible constraint in 'asm'
 #define QLO2(B, N) asm __volatile__("sbrs %[" #B "], " #N ASM_VARS ); LO1;
                                                                     ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:298:15: note: in expansion of macro 'QLO2'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
               ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:150:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:152:71: note: in expansion of macro 'LO1'
 #define QLO2(B, N) asm __volatile__("sbrs %[" #B "], " #N ASM_VARS ); LO1;
                                                                       ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:298:15: note: in expansion of macro 'QLO2'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
               ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:205:97: error: impossible constraint in 'asm'
 #define IDATA2 asm __volatile__("add %A[data], %[ADV]\n\tadc %B[data], __zero_reg__"  ASM_VARS );
                                                                                                 ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:298:27: note: in expansion of macro 'IDATA2'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
                           ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:177:47: error: impossible constraint in 'asm'
 #define CLC1 asm __volatile__("clc" ASM_VARS );
                                               ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:298:34: note: in expansion of macro 'CLC1'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
                                  ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:150:110: error: impossible constraint in 'asm'
[... 408 lines cut ...]
 #define QLO2(B, N) asm __volatile__("sbrs %[" #B "], " #N ASM_VARS ); LO1;
                                                                     ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:313:15: note: in expansion of macro 'QLO2'
     HI1 D1(1) QLO2(b2, 0)      D2(0)  LO1     D3(0)
               ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:150:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:152:71: note: in expansion of macro 'LO1'
 #define QLO2(B, N) asm __volatile__("sbrs %[" #B "], " #N ASM_VARS ); LO1;
                                                                       ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:313:15: note: in expansion of macro 'QLO2'
     HI1 D1(1) QLO2(b2, 0)      D2(0)  LO1     D3(0)
               ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:150:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
C:\EE\microController\Arduino\libraries\FastLED/clockless_trinket.h:313:39: note: in expansion of macro 'LO1'
     HI1 D1(1) QLO2(b2, 0)      D2(0)  LO1     D3(0)
                                       ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Zitt, 2015-07-21:
Now that I'm looking closer; I wonder if there is a compile issue because _AVR_ATtiny167__ has no target in fastpin_avr.h. Any thoughts as to how to get started?

focalintent, 2015-07-21:
What target are you selecting for the attiny167 (and/or what needs to be added to a 1.6.5 install to build for the 167?)

Odds are you are seeing this because there are no pin definitions being set for the attiny167 platform, which is messing with the generated asm.  You need to make sure that fastpin_avr.h in platforms/avr has pin definitions for the 167.

Zitt, 2015-07-21:
Focal - that looks to be the problem. no 167 definitions in fastpin_avr.h
Unsure how to add them correctly; any pointers?

focalintent, 2015-07-21:
You want to add a block something along these lines:

```
#elif defined(__AVR_ATtiny167__) 
_IO(A); _IO(B);

_DEFPIN_AVR(0, 0x01, A); _DEFPIN_AVR(1, 0x02, A); _DEFPIN_AVR(2, 0x04, A); _DEFPIN_AVR(3, 0x08, A);
_DEFPIN_AVR(4, 0x10, A); _DEFPIN_AVR(5, 0x20, A); _DEFPIN_AVR(6, 0x40, A); _DEFPIN_AVR(7, 0x80, A);
_DEFPIN_AVR(8, 0x01, B); _DEFPIN_AVR(9, 0x02, B); _DEFPIN_AVR(10, 0x04, B);
_DEFPIN_AVR(11, 0x08, B); _DEFPIN_AVR(12, 0x10, B); _DEFPIN_AVR(13, 0x20, B);
_DEFPIN_AVR(14, 0x40, B); _DEFPIN_AVR(15, 0x80, B);

#define HAS_HARDWARE_PIN_SUPPORT 1
```

focalintent, 2015-07-21:
Or pull the most recent FastLED3.1 branch, where I Just checked in the above code...

closed 2015-07-21 by focalintent