FastLED/FastLED #193: illegal opcode Compile errors w/ Arduino 1.6.5 on AVR tiny167

Zitt, 2015-07-21:
After fixing #192 ... this next compile error showed up:
Arduino: 1.6.5 (Windows 7), Board: "Digispark Pro (Default 16 Mhz)"

Using library FastLED in folder: C:\EE\microController\Arduino\libraries\FastLED (legacy)

C:\EE\microController\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny167 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_DIGISPARKPRO -DARDUINO_ARCH_AVR -IC:\Users\John Zitterkopf\AppData\Roaming\Arduino15\packages\digistump\hardware\avr\1.6.5\cores\pro -IC:\Users\John Zitterkopf\AppData\Roaming\Arduino15\packages\digistump\hardware\avr\1.6.5\variants\pro -IC:\EE\microController\Arduino\libraries\FastLED R:\temp\Zitt_tmp\Local\build3876144464251568424.tmp\Blink.cpp -o R:\temp\Zitt_tmp\Local\build3876144464251568424.tmp\Blink.cpp.o 

R:\temp\Zitt_tmp\Local\ccHwve8c.s: Assembler messages:
R:\temp\Zitt_tmp\Local\ccHwve8c.s:487: Error: illegal opcode mul for mcu avr35
Error compiling.

focalintent, 2015-07-21:
Ah - right, ATTiny167 likely doesn't have a hardware multiply either...  will fix that in a moment.

focalintent, 2015-07-21:
Checked in - re-pull the 3.1 branch.

focalintent, 2015-07-21:
(note: I don't support the hardware spi on the attiny class mcu's at the moment, so using a chipset like the APA102 will use bitbanged spi output).

closed 2015-07-21 by focalintent

focalintent, 2015-07-21:
(ignore that last comment - turns out the ATTiny87/167's hardware SPI looks identical to the atmega328, so i've tentatively turned on support for it - data is on pin 4 (PA4) and clock is on pin 5 (PA5))

Zitt, 2015-07-22:
I can confirm that after applying these changes from 192 and 193 to FastLED 3.0.1... the code compiles and appears to work on the Digispark Pro running on Tiny167. I am NOT using hardware SPI; so it's bitbanging in my case. 

I'm using Pins PA7 (CLK) and PA6 (DATA) because I wanted to leave the programming pins alone for ISP.

focalintent, 2015-07-22:
Sadly, the digispark pro is still using an attiny that doesn't have a hardware multiplier - this is going to make all sorts of math operations much slower :(  I'm disappointed that they stuck with the tiny for the pro.

commit 33e0a80c (2015-07-21): Fix #193 - add ATtiny167 to the platforms marked for the attiny version of lib8tion.