FastLED/FastLED #195: Bug in fill_gradient_RGB, swapping end and start points done incorrectly?

bourkemcrobbo, 2015-08-03:
I was reading through this code at lines 67-76 in colorutils.cpp;
{
// if the points are in the wrong order, straighten them
if( endpos < startpos ) {
    uint16_t t = endpos;
    CRGB tc = endcolor;
    startpos = t;
    startcolor = tc;
    endcolor = startcolor;
    endpos = startpos;
}

and to me it seems to set the start position to the end position, and then set the end position to the start position, which was just set as the end position. Both start and end positions look like they will be set to the same value.

I'm at uni, so can't test this on my hardware. I may have made a mistake in my reading, but I don't think I have

kriegsman, 2015-08-03:
Yep- looks like a plain, old-fashioned Oops.  I'll fix it.

kriegsman, 2015-08-04:
Thanks for the bug report; good spotting.

commit 8223bd5d (2015-08-04): Fix #195 bug with fill_gradient_RGB, also added TProgmemRGBGradientPalettePtr type synonym
closed 2015-08-04 by kriegsman