FastLED/FastLED #196: CHSV issue with brightness <16

atuline, 2015-08-04:
I've provided some code at http://pastebin.com/imFrYj1K

This program demonstrates (to me) that when setting CHSV brightness to less than 16, the LED's remain dark if:

- FastLED.show() is performed only in setup.
- FastLED.show() is performed within EVERY_N_MILLISECONDS loop at ~20 ms.

LED are lit when:

- FastLED.show() performed directly inside loop().

Details of the environment I'm using are in the link.

focalintent, 2015-08-04:
(I'll have to dig deeper into this - but in the meantime, don't call show in setup -- I've seen it cause problems in the past because of interrupt disabling)

kriegsman, 2015-08-04:
Thanks for the report... and I do agree with Dan: avoid calling show() inside setup().

What _numeric_ values are you seeing in the led array, if you print them out?  

Knowing this will help us figure out if the wrong numbers are getting in there, or if the right numbers are getting in there, and they're being displayed incorrectly on the LEDs for some reason.

focalintent, 2015-08-04:
>LED are lit when:
> * FastLED.show() performed directly inside loop().

in my test setup here with an atmega328 this is absolutely _not_ true.  Which is to say, with using a brightness of 15 for your CHSV value, I never get the leds turning on at all.  I think you're in show outside of show vs. EVERY_N_MILLISECONDS is a bit of a red herring.

However, CHSV(0,255,15) is giving me rgb values of (1,0,0) -- which, given what the inline scaling does, i'm not at all surprised that the output is 0 (scale8(1,255) = 0).

atuline, 2015-08-04:
Interestingly, it was when I was using the EVERY_N_MILLISECONDS call that I started seeing it as I normally never run FastLED.show() in setup. Anyways,

when setting CHSV(0,255,15)  then  leds[i].r = 1
when setting CHSV(0,255,16)  then  leds[i].r = 2

I have updated the above link and commented out FastLED.show() from setup. It's still inside the EVERY_N_MILLISECONDS(20) and exhibiting the same issue.

focalintent, 2015-08-04:
The rainbow HSV conversion pre-scales the brightness value by itself (scale8_video(val,val)) as part of making a less linear brightness curve.  since scale8_video(15,15) = 1 - that's going to scale your output values down to a max of 1 when you are using a brightness of 15 for your CHSVs.  (I'll let @kriegsman chime in more about this)

focalintent, 2015-08-04:
(You can verify this yourself by enabling serial output and putting the following code into your loop:

```
    Serial.print(leds[0].r); Serial.print(":");
    Serial.print(leds[0].g); Serial.print(":");
    Serial.println(leds[0].b);
```

you will see line upon line of 1,0,0 -- which is going to output as black.

atuline, 2015-08-04:
Indeed, my understanding is that an CRGB(1,0,0) should be black, however one would not think that CHSV(0,255,15) would scale to CRGB(1,0,0).

What's also interesting to me is that CHSV(0,255,15) does show up when it's directly within loop.

kriegsman, 2015-08-04:
Thanks for the numbers. You wrote that
  CHSV(0,255,15) ... leds[i].r = 1
  CHSV(0,255,16) ... leds[i].r = 2

And this part seems basically 'working as intended'.

But then here where the trouble starts, and you've got two options, which I'll get to in a moment.  When a value of "1" is sent to FastLED.show(), we use 'scale8( ledValue, masterBrightness)' to affect the master brightness control that you get with setBrightness.  And the problem is that scale8( 1, 255) rounds down to zero, and you get no light.

Obviously this isn't what you want.  Or us.  We don't want this either.

Here are your two choices: (1) somehow make sure that the background pixel values are at least "2" if you want them lit, and there are a bunch of ways to do that; that's not rocket science, and besides, you're one of the world's leading FastLED practitioners.  The other option is (2) wait.  Wait until Dan and I roll out the 'new and improved scale8 asm code' as described here https://plus.google.com/u/1/102282558639672545743/posts/P8pqxywsdgV just a few days ago.  And since we don't have a specific timeline for that, it's hard to tell how long the wait will be.  

I think you're better off writing a little filter function that adjusts pixel values from "1" to "2" in each channel.  Just
  if( pixel.r == 1) pixel.r = 2;
  if( pixel.g == 1) pixel.g = 2;
  if( pixel.b == 1) pixel.b = 2;

And in the glorious future when we roll out the new scale8 asm code, this problem will go away, and you can just delete your pixel adjustment code.

Sorry that it isn't working as expected; we really want to 'fix' the 'scale8' issues, and we're just shorter on hours to put into it all than we'd prefer.

focalintent, 2015-08-04:
Also, here are the first 64 bits of output from CRGB(CHSV(0,255,i)) where i'm cycling i from 0-255:
```
0 == (0,0,0)
1 == (1,0,0)
2 == (1,0,0)
3 == (1,0,0)
4 == (1,0,0)
5 == (1,0,0)
6 == (1,0,0)
7 == (1,0,0)
8 == (1,0,0)
9 == (1,0,0)
10 == (1,0,0)
11 == (1,0,0)
12 == (1,0,0)
13 == (1,0,0)
14 == (1,0,0)
15 == (1,0,0)
16 == (2,0,0)
17 == (2,0,0)
18 == (2,0,0)
19 == (2,0,0)
20 == (2,0,0)
21 == (2,0,0)
22 == (2,0,0)
23 == (3,0,0)
24 == (3,0,0)
25 == (3,0,0)
26 == (3,0,0)
27 == (3,0,0)
28 == (4,0,0)
29 == (4,0,0)
30 == (4,0,0)
31 == (4,0,0)
32 == (5,0,0)
33 == (5,0,0)
34 == (5,0,0)
35 == (5,0,0)
36 == (6,0,0)
37 == (6,0,0)
38 == (6,0,0)
39 == (6,0,0)
40 == (7,0,0)
41 == (7,0,0)
42 == (7,0,0)
43 == (8,0,0)
44 == (8,0,0)
45 == (8,0,0)
46 == (9,0,0)
47 == (9,0,0)
48 == (10,0,0)
49 == (10,0,0)
50 == (10,0,0)
51 == (11,0,0)
52 == (11,0,0)
53 == (11,0,0)
54 == (12,0,0)
55 == (12,0,0)
56 == (13,0,0)
57 == (13,0,0)
58 == (14,0,0)
59 == (14,0,0)
60 == (15,0,0)
61 == (15,0,0)
62 == (16,0,0)
63 == (16,0,0)
64 == (17,0,0)
```

focalintent, 2015-08-04:
(Above gotten via:)

```C++
#include <FastLED.h>
void setup() { Serial.begin(57600); }

void printRGB(const CRGB & rgb) {
  Serial.print("(");
  Serial.print(rgb.r); Serial.print(",");
  Serial.print(rgb.g); Serial.print(",");
  Serial.print(rgb.b); Serial.print(")");
}

void loop() {
  for(int i = 0; i < 255; i++) {
    Serial.print(i); Serial.print(" == "); printRGB(CRGB(CHSV(0,255,i))); Serial.println("");
  }
  delay(10000);
}
```

focalintent, 2015-10-03:
Closing this, as this is hsv2rgb's gamma correction working as intended.  (Fixing scale8 is a separate issue which is a larger scope to fix than this ticket).

closed 2015-10-03 by focalintent