FastLED/FastLED #197: 6 strips of WS2812 on seperate pins. Black becomes green

metri666, 2015-08-07:
I made a circuit with an Atmega 328p. I use 6 pins for different strips;A0, A1, A2, A3, A4, 2. With any 5 pins, everything is OK, but as soon as I add the 6th, black becomes green. So no way to turn the LEDs off. I can still set colors and so simple animations but no Black....

I tried all strips using single Arrays, multi dimensional array & all strips using the same array data. 5 strips OK, but with 6 black becomes green. I tried pins 10, 9, 8 as alternates to any of the other pins with no effect. 5 is always OK, 6 is never OK.

focalintent, 2015-08-07:
What version of the arduino environment and/or what version of avr-gcc are you using?  (Also, what branch?)

metri666, 2015-08-08:
Arduino 1.6.3
FastLED 3.0.3 Branch
I'm really sorry if I should be using 3.1 Branch. I can test that tonight too.

metri666, 2015-08-08:
Fixed in 3.1 Branch.

closed 2015-08-08 by metri666

Nagymadar, 2019-07-15:
It is still an issue it seems. I have the same on ESP-WROOM-32. Adding a 6th strips brakes the program completely. It starts something, but then freezes after ~100 milliseconds or does some random stuff. I have a 8x 272 LED strips on 8 pins. I reduced the length of the strips. No effect. Separate arrays instead of array of an array. Same. I have everything the latest. Any idea?