FastLED/FastLED #199: Build errors with attiny

einsteinx2, 2015-08-11:
I installed the attiny board files using the board manager from https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json

When I compile, I get a ton of errors with macros:

     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                           ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:209:103: error: impossible constraint in 'asm'
 #define SCALE22(B, N) asm __volatile__("sbrc %[s2], " #N "\n\tadd %[" #B "], %[scale_base]" ASM_VARS );
                                                                                                       ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:232:36: note: in expansion of macro 'SCALE22'
 #define RORSC24(B, N) ROR1(B) CLC1 SCALE22(B, N)
                                    ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:364:27: note: in expansion of macro 'RORSC24'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                           ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:199:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:364:48: note: in expansion of macro 'LO1'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                                                ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:224:60: error: impossible constraint in 'asm'
 #define ROR1(B) asm __volatile__("ror %[" #B "]" ASM_VARS );
                                                            ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:364:52: note: in expansion of macro 'ROR1'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                                                    ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:226:47: error: impossible constraint in 'asm'
 #define CLC1 asm __volatile__("clc" ASM_VARS );
                                               ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:364:61: note: in expansion of macro 'CLC1'
     HI1 D1(1) QLO2(b1, 1) RORSC24(b2,7)  D2(4) LO1 ROR1(b2) CLC1 D3(2)
                                                             ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:197:110: error: impossible constraint in 'asm'
 #define HI1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[hi]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=hi; }
                                                                                                              ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:365:5: note: in expansion of macro 'HI1'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
     ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:201:69: error: impossible constraint in 'asm'
 #define QLO2(B, N) asm __volatile__("sbrs %[" #B "], " #N ASM_VARS ); LO1;
                                                                     ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:365:15: note: in expansion of macro 'QLO2'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
               ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:199:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:201:71: note: in expansion of macro 'LO1'
 #define QLO2(B, N) asm __volatile__("sbrs %[" #B "], " #N ASM_VARS ); LO1;
                                                                       ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:365:15: note: in expansion of macro 'QLO2'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
               ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:254:97: error: impossible constraint in 'asm'
 #define IDATA2 asm __volatile__("add %A[data], %[ADV]\n\tadc %B[data], __zero_reg__"  ASM_VARS );
                                                                                                 ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:365:27: note: in expansion of macro 'IDATA2'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
                           ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:226:47: error: impossible constraint in 'asm'
 #define CLC1 asm __volatile__("clc" ASM_VARS );
                                               ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:365:34: note: in expansion of macro 'CLC1'
     HI1 D1(1) QLO2(b1, 0) IDATA2 CLC1  D2(3)  LO1     D3(0)
                                  ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:199:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:365:47: note: in expansion of macro 'LO1'
[... 404 lines cut ...]
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:201:69: error: impossible constraint in 'asm'
 #define QLO2(B, N) asm __volatile__("sbrs %[" #B "], " #N ASM_VARS ); LO1;
                                                                     ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:380:15: note: in expansion of macro 'QLO2'
     HI1 D1(1) QLO2(b2, 0)      D2(0)  LO1
               ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:199:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:201:71: note: in expansion of macro 'LO1'
 #define QLO2(B, N) asm __volatile__("sbrs %[" #B "], " #N ASM_VARS ); LO1;
                                                                       ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:380:15: note: in expansion of macro 'QLO2'
     HI1 D1(1) QLO2(b2, 0)      D2(0)  LO1
               ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:199:110: error: impossible constraint in 'asm'
 #define LO1 if((int)(FastPin<DATA_PIN>::port())-0x20 < 64) { asm __volatile__("out %[PORT], %[lo]" ASM_VARS ); } else { *FastPin<DATA_PIN>::port()=lo; }
                                                                                                              ^
/Users/bbaron/Documents/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:380:39: note: in expansion of macro 'LO1'
     HI1 D1(1) QLO2(b2, 0)      D2(0)  LO1
                                       ^
Error compiling.

Sketch uses 6,746 bytes (23%) of program storage space. Maximum is 28,672 bytes.
Global variables use 244 bytes (9%) of dynamic memory, leaving 2,316 bytes for local variables. Maximum is 2,560 bytes.

focalintent, 2015-08-11:
The only really tested version of the avr code that I support is for the trinket and the gemma.  Specifically, which attiny platform are you trying to compile for with that build?

einsteinx2, 2015-08-11:
Wow thanks for the instant reply! I'm trying to use a bare ATTiny85 chip (same chip in the Trinket). I installed the board files and then chose the ATTiny85 8MHz internal clock board. I'm using FastLED 3.1.

Based on the comments in https://github.com/FastLED/FastLED/issues/192 it seems like there is potentially a simple fix?

focalintent, 2015-08-11:
Using Arduino 1.6.5, with the latest pull from the link you gave above as well as the most recent FastLED 3.1 branch, I'm able to compile code for the ATTiny85 w/8Mhz internal clock board without any issues.  What is the full code that you're trying to compile (either gist or pastebin link, or wrap it in ``` tags so that the code is readable if you put it in a comment here)?

einsteinx2, 2015-08-11:
Here's the code I'm trying to compile: https://gist.github.com/einsteinx2/bae819a563859b3e2e6a

I'm using the latest version of Arduino 1.6.5.

focalintent, 2015-08-11:
(note: you will get errors like that if you try compiling for an impossible pin - and with the attiny's, the ATTiny85 only has 6 pins defined - you're using pins 9, 10, 11, and 12 -- none of which are valid pins for the ATTiny85)

einsteinx2, 2015-08-11:
THANKS!! I had started working on this project with the idea to use pro mini's and then decided to switch to the ATTiny85 later since I had extras laying around and forgot to change the pins!

closed 2015-08-11 by einsteinx2