FastLED/FastLED #200: fill_gradient* looses original value of startpos if endpos<startpos

Spirou42, 2015-08-17:
Hi, i stumbled upon the start- and endpos/color swapping in fill_gradient*. The current implementation 
```
if( endpos < startpos ) {
        uint16_t t = endpos;
        CHSV tc = endcolor;
        startpos = t;
        startcolor = tc;
        endcolor = startcolor;
        endpos = startpos;
    }
```
sets *startpos* to *endpos* and afterwards *endpos* to *startpos* loosing the original value of *startpos*.

This version
```
if( endpos < startpos ) {
        uint16_t t = endpos;
        endpos = startpos;
        startpos = t;

        CRGB tc = endcolor;
        endcolor = startcolor;
        startcolor = tc;
    }
```
works better.

focalintent, 2015-08-17:
Already fixed on the 3.1 branch.

Spirou42, 2015-08-18:
Unfortunately it is not.
If i check the current version of colorutils.h on the FastLED 3.1 branch, i find the exact same faulty code in the fill_gradient template class and also in the fill_gradient_RGB function in colorutils.cpp.

kriegsman, 2015-08-18:
Thanks- I'll check those out and fix ASAP as needed.

kriegsman, 2015-08-18:
The template code in colorutils.h did have that bug still, thank you.

I'm not seeing the problem in colorutils.cpp on the 3.1 branch; I believe that was fixed about two weeks ago in https://github.com/FastLED/FastLED/commit/8223bd5d7e7a01dfc6d17f36889853e046d7f47e Double-check that you have the latest 3.1 branch checked out, and please do let me know if you're still seeing a problem.

closed 2015-08-18 by kriegsman

Spirou42, 2015-08-18:
My fault, the *fill_gradient_RGB* function on the FastLED3.1 branch is Ok.

kriegsman, 2015-08-18:
No problem! Thanks for the ticket!