FastLED/FastLED #203: Support for TLC5940

pcharest2000, 2015-08-24:
This chip is readily available and cheap on ebay.  I know this is a 12 bit pwm IC. But please find attached my post on a thread regarding this request i.e the chip can be driven in 8 bit pwm thus needing no change to the core of fastled.  I would love do contribute to the effort and provide the "driver" for AVR.  On ARM cortext (M3) I am using DMA to drive this IC....  

focalintent and kriegsman: The tlc5940 can be used in 8 bit without mapping to 16 bit. The trick is to fill the 12 bit data with 8 bit pwm value and then trigger the blank signal when you have saturated the 8 bit value. I have done this on several of my project on Avr and Arm cortex. This consumes very little resources since the GSCLK and Blank signal are all generated thru PWM. Also using SPI on AVR when sending one byte you have about 18 cycles between SPI send. This gives just enough time to built the "12 bit" packed stream between SPI send. The TLC5940 could be supported without touching the base code color functions of FastLed, of course you don't get 12 bit resolution on the TLC 5940 but who really cares? I do not understand why such a high resolution was used for this IC by TI, I don't think the normal human can spot the difference between 8 bit and 12 bit color value....

Phil

focalintent, 2015-08-24:
This is a duplicate of #18 - already tracking for this (in fact your comment that you posted comes from that issue).  Mostly what I need to get the support in is to get some test hardware up and running here and then do the chipset implementation.  Want to accelerate this?  Shoot me a board that's already populated w/leds and a tlc5940 (I've got a bunch of hardware somewhere, but need to get it all soldered together - and so it keeps dropping off of the list of things to work on).

(Also, the library already supports a couple of 12-bit chipsets w/8-bit data, I've already gotten that running).

closed 2015-08-24 by focalintent

pcharest2000, 2015-08-25:
Daniel,

Please PM at philippedotcharest@gmail.com and send me your information and
I will happily send you a dirty "protoboard"  with 2 TLC5940 soldered o it
n with the different signal lines...  if this help contributes to your
great library.  If you feel there is not a market for such a chip you  can
drop it,  my point was being this chip can be run in 8 bit mode.

Phil

focalintent, 2015-08-25:
Sent!  Thanks -- mostly the chip has remained a low priority because so few people are using it (the primary emphasis these days is on the chips that are used in the led strips/boards that many people are using, so the WS2812 and the APA102 most recently).  I'd love to get this one off of my plate, though :)

Also - there's another benefit to the 12-bit chips.  Right now, when global scaling is used, as a brightness value gets scaled down, the resolution gets cut off.  However, with the 12 bit chipsets, even as the global brightness is scaled down, because it'll fill into the low 4 bits of the 12-bit value going out, you will still retain some level of dynamic range.  So, there is some good reason for the chpis.

Honestly, I'm surprised no one has done a 12 or 16 bit version of the APA102 or WS2812 yet.