FastLED/FastLED #205: Strange problem with the Octows2811demo sketch, teensy 3.1, fastled 31

farleysynth, 2015-09-04:
I have a strange problem with the Octows2811demo sketch, teensy 3.1, fastled 31.

If NUM_LEDS_PER_STRIP  is 864 or less, it works.  if 865 or greater, it hangs.  Paul's Octows2811 demo sketches ( fire, rainbow) work fine with 900 LEDS/strip.

Can someone please run this sketch?  Do you have the same experience?  Does it work with 865? 900?

You do not need LEDs hooked up.  In fact, you can comment out the LED loading for loops.  I've tried 2 different teensy 3.1

PEBKAC ???   Weird memory overflow problem?  

THANKS!  Please let me know if you need more information or have some suggestions.   

```
#define USE_OCTOWS2811

#include<OctoWS2811.h>

#include<FastLED.h>

#define NUM_LEDS_PER_STRIP 864
#define NUM_STRIPS 8

CRGB leds[NUM_STRIPS * NUM_LEDS_PER_STRIP];

// Pin layouts on the teensy 3:
// OctoWS2811: 2,14,7,8,6,20,21,5

void setup() {
  delay(5000);
  Serial.begin(9600);
  LEDS.addLeds<OCTOWS2811>(leds, NUM_LEDS_PER_STRIP);
//  LEDS.setBrightness(32);
}

void loop() {
  static uint8_t hue = 0;
  for(int i = 0; i < NUM_STRIPS; i++) {
    for(int j = 0; j < NUM_LEDS_PER_STRIP; j++) {
      leds[(i*NUM_LEDS_PER_STRIP) + j] = CHSV((32*i) + hue+j,192,255);
    }
  }

  // Set the first n leds on each strip to show which strip it is
  for(int i = 0; i < NUM_STRIPS; i++) {
    for(int j = 0; j <= i; j++) {
      leds[(i*NUM_LEDS_PER_STRIP) + j] = CRGB::Red;
    }
  }

  hue++;
  Serial.println("To LEDS.show");
  LEDS.show();
  Serial.println("Back from LEDS.show");
  LEDS.delay(100);
 
}
```

focalintent, 2015-09-04:
It is because it I've chosen to prioritize performance over memory efficiency.  Because of how octows2811 works, and how FastLED works, there a three large blocks of memory that get allocated. 

The first is your LEDs array - which is going to be over 20k (864 * 8 * 3). 

The second is the buffer that octows2811 is writing out of, which is the same size, and will also be 20k. 

Finally, there is a third buffer that is used to translate/scale the first buffer while octows2811 may still be writing data out of the second buffer.  Without this third buffer, I would have to wait to begin scaling and bit rotating the LEDs array into the format octows2811 wants until it was _completely_ finished writing the previous frame out. 

This puts you at well over 60k of memory usage - just for led data, on a device with only 64k of ram, and because two of those buffers are allocated by malloc, it is probably unable to make them fit, and so hangs on the second malloc.

farleysynth, 2015-09-04:
PEBKAC.  How obvious.  I'd seen the two buffer mallocs but didn't put it all together.  Thank you for taking the time to write the informative explanation.    Thanks for the impressive and fun library, and thank you for sharing you work and guidance.

closed 2015-09-04 by farleysynth