FastLED/FastLED #206: Arduino 1.6.5 Support

W7PEA, 2015-09-06:
Please catch up to the latest Arduino version.

```
In file included from C:\Program Files (x86)\Arduino\libraries\FastLED/controller.h:4:0,
                 from C:\Program Files (x86)\Arduino\libraries\FastLED/FastLED.h:27,
                 from ButtonExample.ino:1:
C:\Program Files (x86)\Arduino\libraries\FastLED/led_sysdefs.h:38:2: error: #error Arduion versions 1.5.7 and later not yet supported by FastLED for AVR
 #error Arduion versions 1.5.7 and later not yet supported by FastLED for AVR
  ^
ButtonExample.ino:2:63: fatal error: Button.h: No such file or directory
compilation terminated.
Error compiling.
```

kasperkamperman, 2015-09-06:
Do you use the FastLED 3.1 branch? I work with Arduino 1.6.5 and FastLED for a long time without problems.

NicoHood, 2015-09-06:
Seems that its time to push 3.1 to avoid those issues. Imagine of the people who dont have a github account, dont report this "issue" and give up or just use the adafruit library which is.. you all know.

kriegsman, 2015-09-06:
Oh believe me, we're well aware of the problem this causes!  We've been closing in on a full 3.1 release a lot recently. Stay tuned.

closed 2015-09-06 by kriegsman

W7PEA, 2015-09-07:
didn't know there was a 3.1 in the works. Was sent here from a link on BlinkLabs site. I'll grab 3.1, thanks!

Link for the unaware like me : https://github.com/FastLED/FastLED/tree/FastLED3.1