FastLED/FastLED #209: FastLED 3.1 and Arduino Mega Issue (Arduino Leonardo works)

kennyviperhk, 2015-09-08:
#include "FastLED.h"

#define NUM_LEDS_1 350

#define DATA_PIN_1 10

// Define the array of leds
CRGB leds_1[NUM_LEDS_1];

int e[4];

void setup() {

  FastLED.addLeds<WS2811, DATA_PIN_1, RGB>(leds_1, NUM_LEDS_1);

  Serial.begin(115200);
  e[0] = 0;
  e[1] = 0;
  e[2] = 0;
  e[3] = 0;
}

void loop() {

  if (Serial.available() > 3)
  {
    for (int i = 0; i < 4 ; i++)
    {
      e[i] = Serial.read();
    }
    Serial.flush();
  }
  leds_1[e[0]].setRGB(e[1], e[2], e[3]);
  FastLED.show();
}

=========
I have an Openframeworks software sending Serial Signal sending bytes of (Position, R, G, B) for controlling the LED array. 
The usb Serial Communication works perfectly on Arduino Leonardo. 
However it doesn't get any signal on Mega 2560. the RX LED blinks but nothing happen. the Mega itself works fine with example code like Blink without Serial.
Also have tested with Mega 1280....the signal would get weird like wrong colours or skipping the signals.

Is there a conflict with the Serial port?

Thanks in advance

focalintent, 2015-09-09:
I suspect what you're running into is a problem caused by the fact that interrupts get disabled when writing out LED data.  With the leonardo, if you're having serial data come in over the USB port then there is a fairly large buffer that can hold data and ensure that you don't lose data during the period of time when interrupts are disabled from writing out led data.  However, with the mega's, there is no such thing, and so while led data is being written out, interrupts are disabled, which means that random amounts of data coming in on the serial port may be lost.

The solution is to either change your leds to ones that don't require disabling interrupts while writing data out (APA102, LPD8806, etc...), or change your protocol to one that isn't attempting to send data over serial while the mega is writing data out to leds (for example, have the mega send an OK to the other side to indicate that it is ok to send data, wait to read all the data from the other side, and only then show led data), or switch to something like the teensy 3/3.x where it is possible to allow limited interrupts to run which would allow the serial input handling to occur without losing data.

closed 2015-09-09 by focalintent

NicoHood, 2015-09-17:
You must use Serial1 for the Serial HW port on the leonardo.

I am doing something very similar (with adalight, via HW and CDC Serial), but this suppressed my plan:
https://github.com/arduino/Arduino/issues/3745

focalintent, 2015-09-17:
@NicoHood you're reading his report backwards - his code is working on the leonardo, not on the mega2560.

NicoHood, 2015-09-17:
Oh, sorry.
Serial.flush is not needed, since you never send any data. Try to leave this out. Maybe thats the problem I linked above. Maybe you set the Serial speed wrong? For the leonardo the speed doesnt matter, but for the Mega.
Also try lower speeds.