FastLED/FastLED #210: Corrupted Colour Space when using more than five FastLED controllers ?

JayMonkey, 2015-09-09:
Hi,

Issue background ....

I'm using FastLED lib with a project that i'm working on. 

Arduino Nano V3 (have also tried with Uno & Mini Pro)
6 x WS2812B LED Strips connected to individual digital I/O ports as follows 

Strip 1 Data Pin = D9
Strip 2 Data Pin = D8
Strip 3 Data Pin = D7
Strip 4 Data Pin = D6
Strip 5 Data Pin = D5
Strip 6 Data Pin = D4

The 5V for all the WS2812b is suppled by an external source and is not using the USB 5V power, but the ground are tied together ....

Basically i want to do is create a separate FastLED controller for each strip, which i've tried to do using the documentation from your git repo here 

https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples

However no matter what method used (2D or 3D array) it seems that defining more than five FastLED controllers using CRGB method causes the colour space array for all leds and controllers to become corrupted ... eg a RGB value of (0,0,0) will show solid green ?, dropping back down to 5 controllers or less and the code runs fine.

Defining more than five FastLED controllers (doesn't matter if its added to the existing array or defined as a new separate controller using a different name) the colour space becomes corrupted ... ?

I'm not sure if this is an issue in the FastLED Library or an issue with the Arduino IDE compiler, either way it looks like some sort of memory allocation error (or pointer ref) is the cause ...

In all cases the code compiles and uploads fine ....  no errors or warnings

The following example code (derived from your doc examples) creates/shows the issue, as it is below it runs fine with five controllers defined, however change the number of strips to 6 and uncomment the 6th controller definition and the colour space goes whack-o ?

#include "FastLED.h"
#define NUM_STRIPS 5    // <-- Change to 6 to see issue
#define NUM_LEDS_PER_STRIP 3

CRGB leds[NUM_STRIPS][NUM_LEDS_PER_STRIP];   // Create Array for Strips & LED Colour space

void setup() {
  FastLED.addLeds<NEOPIXEL, 9>(leds[0], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 8>(leds[1], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 7>(leds[2], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 6>(leds[3], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 5>(leds[4], NUM_LEDS_PER_STRIP);
 //  FastLED.addLeds<NEOPIXEL, 4>(leds[5], NUM_LEDS_PER_STRIP);     // <-- Uncomment for issue

}

void loop() {

// Moving DOT Demo (should show red dot up and down all LEDS & STRIPS)

  for(int x = 0; x < NUM_STRIPS; x++) {
    for(int i = 0; i < NUM_LEDS_PER_STRIP; i++) {
      leds[x][i] = CRGB::Red;
      FastLED.show();
      delay(10);;
      leds[x][i] = CRGB::Black;
      FastLED.show();
      delay(10);
    }
  }
}

Using any of the multi-controller methods shown in the example documentation causes the issue  (if more than five controllers) i've tried different Arduino's and dig I/o port assignments but the issue remans and is constant, in above example once configured for 6 controllers and is showing the issue commenting out say the controller on pin 7 and the colours will be ok again ..

I hope this is enough information for you to understand and recreate the issue, ultimately i would like to have at least 7 controllers all on different dig I/O pins the number of LEDs on each strip is not important it can be 2 or 200 and the issue still occurs.

If you need any further testing or debug from me please just ask ...

Cheers
Jay

focalintent, 2015-09-09:
You haven't mentioned which compiler you are using, but it is most likely an issue with the compiler combined with the fact that you're using something with a relatively low program space size.  First thing is try FastLED3.1 with arduino 1.6.5 or later.  

Secondly - because of how the WS2812 code is written, if you are running into program space limits there's not a lot of options available to you - it is a _lot_ of code, and it is repeated for each port/pin combination - it is possible that when you have more than a handful of instantiations the compiler is trying to be clever and re-use code there, which it shouldn't, it will mess up the timing - either switch to a mega or switch to a teensy 3.x which has no issues with multiple instantiations of the controller (and honestly, with the teensy 3.x you have the option of parallel output, which will improve your performance quite a bit).

JayMonkey, 2015-09-09:
Am using Arduino IDE 1.6.5 will FastLED 3.1 branch ... and try again

Jay

focalintent, 2015-09-09:
In that case, you are most likely hitting a program space limit where the compiler is trying to re-use the template instantiation.   This isn't going to work, though, because as you've seen the compiler will mess it up and throw off the timing for the WS2812's - there isn't a whole lot that I can do to fix this, unfortunately (at least not without a pretty major re-write of the avr asm code which is unlikely to happen, ever) - and so my suggestions above remain - switch to a mega or teensy 3.x.

closed 2015-09-09 by focalintent

JayMonkey, 2015-09-09:
@ focalintent,

Many thanks for your prompt help and advice today ....

I have updated my FastLED libraries to the 3.1 branch and it completely resolves the issue, I can now have more than 5 (tested 7 so far) LED controllers without the issue seen in the FastLED main trunk version ... Happy Days :-)

Is the 3.1 trunk acting as an alpha/beta version in readiness for the next public release ?

Are there any negatives to using the 3.1 branch rather then the master trunk version of FastLED ?

Thanks again
Cheers
Jay

focalintent, 2015-09-09:
No - I have a bunch of housekeeping stuff I need to finish to make the 3.1 branch the master branch - it's just a matter of getting the time : /

JayMonkey, 2015-09-09:
I know the problem well ..... (as in not enough of it)

Nice job with the FastLED libs though dude, seems much more robust with its timings than the neopixels lib which seems to get upset if you try to do anything else clever at the same time ...

Cheers
Jay