FastLED/FastLED #213: Make use of new IDE 1.6.6 features

NicoHood, 2015-09-24:
1.6.6 gives some new options such as:
* c++11
* [.a linkage](https://github.com/arduino/arduino-builder/issues/10)
* [Probably soon a UserSetting.h file](https://github.com/arduino/arduino-builder/issues/15)

You should consider to have a look at this and use it.

For example this can be used:
https://github.com/FastLED/FastLED/blob/f4ddce39bbc7d0e1be89464a13d3cd1d8acb9063/lib8tion.h#L1024-L1030

If you include FastLED.h it adds some general overhead, even if not used. I tried with .a linkage but that did not remove that overhead. However maybe you (who knows the library best) knows how to improve the library with this maybe.

focalintent, 2015-09-24:
I'm already evaluating c++-11 features to add in - but need to be careful about which ones I do/don't both because of potential unintended overheads, but also because of where compiler support is/isn't.  

There's already a number of config type options that someone could put in a user settings type header file - there's even already a single header file in the library that supports it. 

Finally, the nature of the template work FastLED does means that it will never be a static library that you can just link in.

focalintent, 2015-09-24:
Going to close this off as too general.  The C++ features will get added to a list as I review the ones that appear worth adding (and there's a lot that goes into deciding what features to use/not use).

closed 2015-09-24 by focalintent