FastLED/FastLED #215: Compilation error with Arduino 1.6.5 and FastLED 3.1

gomezo76, 2015-10-04:
Hi,

i'm trying to compile PaletteCrossfade sample that needs FastLED3.1, so i updated the lib to the last version but i did not succeed in compiling it :(
here is the compilation error message:

```In file included from PaletteCrossfade.ino:1:0:
/Users/julien/Documents/Arduino/libraries/FastLED/FastLED.h:83:90: error: expected template-name before '<' token
 template<uint8_t DATA_PIN, EOrder RGB_ORDER> class SK6812 : public SK6812Controller800Khz<DATA_PIN, RGB_ORDER> {};
                                                                                          ^
/Users/julien/Documents/Arduino/libraries/FastLED/FastLED.h:83:90: error: expected '{' before '<' token
/Users/julien/Documents/Arduino/libraries/FastLED/FastLED.h:83:90: error: expected unqualified-id before '<' token
Erreur lors de la compilation.```

Thanks for your help

![capture d ecran 2015-10-04 a 15 35 22](https://cloud.githubusercontent.com/assets/10235227/10268078/9153a67a-6aad-11e5-8e9a-6f311917b2ff.png)

focalintent, 2015-10-04:
Pull again, I checked in a fix last night - also use the master branch, now that 3.1 has been released.

gomezo76, 2015-10-04:
Thanks it works now!

closed 2015-10-04 by gomezo76