FastLED/FastLED #217: Multiple controllers

G33kLabs, 2015-10-08:
Hi !

I'm just trying to remove an arduino from my projects and I want to use Neopixel and Dotstar strips on the arduino nano...

After some tests, it looks not possible as it works only on the first declared controller :

```
  // Init touch LEDS
  LEDS.addLeds<NEOPIXEL, TOUCH_LED_PIN>(touchManager.leds, touchManager.NUM_LEDS);
  
  // Init Face LEDS
  LEDS.addLeds<APA102, FACE_DATA_PIN, FACE_CLOCK_PIN, BGR, DATA_RATE_MHZ(12)>(faceManager.leds, faceManager.NUM_LEDS);
```

It works great with one or the other commented.

Is that simply possible ?

Thanks for your answers !

focalintent, 2015-10-08:
This is a thing that I do quite often.  (One of my testing rigs involves multiple types of leds setup at the same time with something that uses multiple controllers like this).  I'll double check to make sure that this is still working, but it was the last time I ran with it.

focalintent, 2015-10-08:
Yeah, this is working here:

```
#include <FastLED.h>

#define NUM_LEDS 96
CRGB leds[NUM_LEDS];
CRGB leds2[NUM_LEDS];

void setup() { 
  FastLED.addLeds<NEOPIXEL,5>(leds, NUM_LEDS);
  FastLED.addLeds<APA102,BGR>(leds2, NUM_LEDS);
}

void loop() { 
  static uint8_t b = 0;
  nscale8(leds, NUM_LEDS, 192);
  nscale8(leds2, NUM_LEDS, 192);
  leds[b] = CRGB::Red;
  leds2[b] = CRGB::Blue;
  ++b %= NUM_LEDS;
  LEDS.show();
  delay(33);
}
```

So - some more questions - what output are you seeing?  Also - can I see the entirety of your code/sketch? I wonder if there's something else that's going on that's being triggered by both sets being active.

G33kLabs, 2015-10-08:
You're right @focalintent !
It works perfectly : count for dotstar leds was too high ! 

Thanks for you help !

closed 2015-10-08 by G33kLabs