FastLED/FastLED #220: Bare AVR support

dolence, 2015-10-21:
I just read about this project and it looks awesome. Would be great if I could use it with a cheap bare AVR microcontroller without having to use the entire Arduino bootloader and libs.
I have a project using an AT90USB162 and would be glad to test it in this way. Or, if someone could kindly give me some directions I could try to port it by myself.
Thansk in advance.

focalintent, 2015-10-21:
Ask on the g+ community - there are some folks there who have gotten the library working directly with AVR studio - http://fastled.io/+ - (i'll keep this open but given the number of platforms that I still need to do support for, as well as major new features coming down the line, this is unlikely to get directly tackled by the library any time soon).

dolence, 2015-10-21:
Thanks for your reply.
I managed to compile using AS7 adding some missing headers from Arduino. Unfortuantely, I didn't tested if it works yet, but will do it ASAP. If one could manage the code to be compiled fine without those Arduino headers I would be glad to contribute with as many AVR pin definition as possible.
Thank you again.

eltang, 2016-11-19:
I would also like to have this.

kriegsman, 2016-11-19:
It's not too far from being able to run on bare AVR already; the core library can exist without "Arduino" functions.

Anyway, it's not a development priority for _us_ right now, but if you get it working on bare AVR, please do submit a pull request, and we'll definitely look at incorporating your changes into the library.

eltang, 2016-11-19:
What exactly are your priorities right now?

kriegsman, 2016-11-19:
I think I'd say that RGBW, support for 'new boards', and general bug fixes are the top items.

However, @focalintent and I just had death in the immediate family, so no library work right now.

kriegsman, 2020-02-17:
Closing this for cleanup of old issues that aren't a current priority.

closed 2020-02-17 by kriegsman