FastLED/FastLED #223: Reversing the order of a Strip

mege, 2015-10-29:
Greetings,

when using an array of LEDS - https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples#array-of-led-arrays - it would be nice to be able reverse the numerical order of a strip.  

For example with two strips, sometimes the strips are placed starting-end to starting-end and radiate outward, other times they will be placed serially ending-end to starting-end.  For example: when using sinelon() it looks odd to have the dot leave one end of a strip and come out on the far end of the other stip.

I spent about an hour looking at the documention and examples but was not able to find any setting that did this.

Am I missing something or is this a new feature request?

Thanks.

marmilicious, 2015-10-29:
Currently there's no FastLED function for reversing the pixels in a strip.  There was a G+ post asking about this awhile back here:
https://plus.google.com/+RaphaelOuzan/posts/DrRjnXtEhjN

marmilicious, 2015-10-29:
See this post also:
https://plus.google.com/112150853881388332803/posts/JgRWYHA7RM8

NicoHood, 2015-10-29:
I currently cannot read the google posts (blocked here) but since we use int as led length in most functions, we could use a negative number to say it should be inverted? Not sure how much overhead this gives. Also not sure if its really needed. But this way you can reverse parts of the strip and not the whole strip. Reversing the whole strip could be easier I guess.

mege, 2015-10-29:
marmilicious: Thanks for the other references.

NicoHood:  Yes, we are on the same page.  It seems like it would be pretty easy to put another variable in the addled constructor, something like LED_STRIP_ORDER.  The default value could be +1 and to reverse the order the value would be -1.

The usesage could look like this to reverse the ordering
#define LED_STRIP_ORDER -1
FastLED.addLeds<NEOPIXEL, DATA_PIN, LED_STRIP_ORDER >(leds, NUM_LEDS_PER_STRIP);
for(int i=0,i<=NUM_LEDS_PER_STRIP, LED_STRIP_ORDER * i++){ ...existing FASTLed code...}

Question: I'm not exactly sure which FASTLed files would need to be changed. Anyone have suggestions for where to start?

komby, 2015-10-29:
> FastLED.addLeds(leds, NUM_LEDS_PER_STRIP);

I do this in my code by populating the data backwards when I am putting
values in the array prior to calling show

if you look towards the end of this sketch
https://github.com/komby/RFShowControl/blob/master/examples/RF_In_Pixel_Out/RF_In_Pixel_Out.ino

Its not the most efficient way to do it but it works for me :)  I am also
supporting grouping of pixels so the loop is a little more complicated than
just reversing the order of the data.

http://www.komby.com

closed 2015-12-07 by focalintent