FastLED/FastLED #224: Multiple strips won't display without creating a single strip too

mege, 2015-10-29:
I have encountered something really odd, I can not get a multiple strip setup to display properly without also creating a controller for a single strip too.

```
//Create three multistrip controllers from six physical led strips
  FastLED.addLeds<LED_TYPE, 2, COLOR_ORDER>(BONDED_STRIPS[0], 0, NUM_LEDS_PER_STRIP[0]);
  FastLED.addLeds<LED_TYPE, 8, COLOR_ORDER>(BONDED_STRIPS[0], NUM_LEDS_PER_STRIP[0], NUM_LEDS_PER_STRIP[1]);
  FastLED.addLeds<LED_TYPE, 6, COLOR_ORDER>(BONDED_STRIPS[1], 0, NUM_LEDS_PER_STRIP[2]);
  FastLED.addLeds<LED_TYPE, 4, COLOR_ORDER>(BONDED_STRIPS[1], NUM_LEDS_PER_STRIP[2], NUM_LEDS_PER_STRIP[3]);
  FastLED.addLeds<LED_TYPE, 10, COLOR_ORDER>(BONDED_STRIPS[2], 0, NUM_LEDS_PER_STRIP[4]);
  FastLED.addLeds<LED_TYPE, 12, COLOR_ORDER>(BONDED_STRIPS[2], NUM_LEDS_PER_STRIP[4], NUM_LEDS_PER_STRIP[5]);
  
//Create a single strip controller.
//The won't work without this line even though the controller is never used elsewhere in the code.
  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorre ction(TypicalLEDStrip); 

```

Has anyone else experienced this?

Suggestions welcomed for where to post a full sketch.

focalintent, 2015-10-29:
Use gist.github.com to post the full code - this is something that I do all the time in my code and i never have to add the extra controller like you are doing there.

focalintent, 2015-12-07:
Are you still having a problem here?  If so - post the entirety of your code to gist.github.com - put the link here and re-open the issue.

closed 2015-12-07 by focalintent