FastLED/FastLED #225: Inconsistent colors produced on TM1803

vinayjain404, 2015-10-30:
I am seeing a varying color when I transition between Red and Green every 10 seconds. I am using Arduino Yun 1.6.5 and using Radio Shack TM1803 TriColor Multistrip.

Below is my code. Main issue is there is a 2/5 chance that the transition color has neither green or red.
How can I fix this issue ?

Thanks for help and BTW this library is awesome to work with.

``` C
#include <ArduinoJson.h>
#include <Bridge.h>
#include <FastLED.h>
#include <HttpClient.h>

#define NUM_LEDS 10

// Data pin that led data will be written out over
#define DATA_PIN 12

CRGB leds[NUM_LEDS];

void setup() {
  // sanity check delay - allows reprogramming if accidently blowing power w/leds
  delay(2000);

  // Initialize the FastLED setup with correct Chipset Name and Pin settings
  FastLED.addLeds<TM1803, DATA_PIN, RGB>(leds, NUM_LEDS);

  // Initialize the bridge so we can use HTTP libraries
  Bridge.begin();

  // Start serial port on freq 9600
  Serial.begin(9600);

  // Wait for serial port to initialize
  while(!Serial);
  Serial.println("setup completed");
}

void loop() {
   HttpClient client;
   StaticJsonBuffer<200> jsonBuffer;
   String response;

   // API Call
   client.get(SAMPLE_API_CALL);
   Serial.println("Making API call");

   // Fetch response
   while (client.available()) {
    response = client.readString();
   }
   Serial.println("Response recevied");
   Serial.flush();

   // JSON decode the object
   JsonObject& root = jsonBuffer.parseObject(response);
   double val = random(2);

   // Vary the color of the LED lights based on the response
   if (val == 0) {
     Serial.println("Display Red");
     FastLED.showColor(CRGB::Red, 255);
   } else {
     Serial.println("Display Green");
     FastLED.showColor(CRGB::Blue, 255);
   }

   Serial.print("Delta of revenue: ");
   Serial.println(val);

   // Wait for 1 min before querying the API
   delay(10000);
} 
```

focalintent, 2015-10-30:
What version of the library are you using?  Make sure you have the most recent, there have been a number of fixes for arduino 1.6.5.  (Also, as a side note:  be aware that FastLED has to disable interrupts when writing out led data for 3-wire led strips like the TM1803 - that could result in losing data over the serial port).

vinayjain404, 2015-10-30:
I am using version 3.1 of the library. How do I disable interrupts before
writing out led data ?

Thanks for the help.

focalintent, 2015-10-30:
> How do I disable interrupts before writing out led data ?

You don't, the library does - however, disabled interrupts may mean that serial data gets lost. 

From your code above it looks like you're just toggling between red and green, right?  - so the colors should just be one or the other.  If that's not what you're seeing - I'm not 100% sure what to recommend.  The TM1803 is one of the worst led chipsets out there (made worse by the fact that there appear to be multiple TM1803 variants, each with different timing - radio shack deserved to go under for selling the TM1803).

Double check your wiring though, make sure that ground is going to your power supply's ground as well as ground on the arduino.  

Also, in your code you are saying "Display Green" - but you are calling ShowColor with ```CRGB::Blue```.  If you need to change your RGB color ordering - look here - https://github.com/FastLED/FastLED/wiki/Rgb-calibration

closed 2015-12-07 by focalintent