FastLED/FastLED #226: Using with mozzi synthesis library?

bnolan, 2015-11-18:
Is it possible to make fastled and mozzi work together? The only way I can make it work at the moment is by fading the synthesis sound to zero, then pausing mozzi while I call FastLED.show(). Are there any techniques I could maybe look into that would enable the two libraries to work together without pausing mozzis use of a timer to create PWM sound?

focalintent, 2015-11-18:
What LEDs are you using?

focalintent, 2015-11-18:
If you are using neopixel/ws2811 style LEDs that's going to be the source of problems, as they require disabling interrupts while writing out led data. 

Switch to an led chipset that doesn't require disabling of interrupts like the apa102 or lpd8806.

bnolan, 2015-11-18:
Yeah I'm using ws2812. So it's just not possible to fix this? :(

focalintent, 2015-11-18:
Not with WS2812's and an uno/nano.  You might be able to make it work on the teensy 3.2 - as there is the option for allowing interrupts to run there, however that requires that Mozzi's interrupt handler be very fast/efficient, so without trying it out, I can't tell you in advance whether or not it will work.

bnolan, 2015-11-18:
Ah I have some lpd8806! I'll try use them thank you. :)