FastLED/FastLED #232: Serial data got lost

Diaoul, 2015-12-03:
I'm using a FastLED sample with NEOPIXEL and some Serial I/O (to talk to ESP8266 as it's not supported by FastLED yet). I've noticed that on high Serial speed, data got lost.
I'm surprise such a big limitation (cannot use straightforward Serial with FastLED) isn't clearly mentionned in the docs somewhere.
I've found #209 and https://github.com/FastLED/FastLED/wiki/Platform-limitations which does not answer all my questions so here is my issue :)

1. Is it possible to suppress this data loss? If yes, what are the consequences?
2. I cannot reproduce the data loss at 9600 baudrate but it doesn't mean it cannot happen right? Or is 9600 baudrate not using interrupts?
3. I'm prototyping on Arduino UNO, will it be the same on a Pro Mini (I assume it will)?

I would like establishing some kind of Serial protocol for between the ESP and the Arduino to be the last resort as it implies another layer of buffering and I don't want to crowed my ESP RAM and keep overhead to a minimum. Also I'd like to keep the price low so Teensy (and better hardware in general) isn't really an option here.

Thanks for your great lib :+1:

focalintent, 2015-12-03:
The chipset page (https://github.com/FastLED/FastLED/wiki/Chipset-reference) mentions WS2811 and variants requiring the disabling of interrupts[1].  And sadly, it isn't possible to suppress it.  

To answer your questions specifically:

1) No, it is not - at least not on a AVR (and I tried, very very hard, to port the interrupt enabling work that I did for ARM back to AVR).
2) So, at 9600 baud, it takes 833µs to transmit a single byte, which means the interrupt handler to read an incoming serial byte needs to be able to fire once every 833µs (with some ability to slide around) to not lose bytes.  At 30µs to write a single WS2811's worth of data, once you get above 35ish leds, you increase the odds of missing an interrupt.  (Realistically, you can probably go higher, maybe even as high as 70 and luck out).
3) Yes, the problem is with AVR - so anything that uses an 8-bit AVR chip (atmega32u4, atmega328p, atmega2560, etc...) will have this problem.

Unfortunately - if you're going to use WS2811's and AVR you're going to need some kind of protocol between the ESP and the Arduino.  The ESP has 96kb of ram, though - nearly 50 times as much as the Arduino - so don't worry about eating up ram on the ESP.

So - what i've recommended to people regularly in tickets and on the g+ group - is basically a simple protocol.  If you can spare an extra pin a simple one would be to simply raise that pin high on the arduino when you are _not_ writing out led data.  Then, on the ESP side you can, when you want to send data, you just check the state of that pin before sending each byte.  

Another alternative would be having the arduino send, via serial, a command to the ESP that says "send me a block of data!" and then read all the data in one chunk before the next round of writing out LED data.

(Yet another alternative would be switching to an LED chipset that doesn't suck as much as the neopixels like the APA102 that don't have interrupt disabling timing requirements :)

On the long backlog of things to do if I ever get time is a bunch of examples showing off how to do serial type stuff w/FastLED and crippled chipsets like the NEOPIXEL and friends :/ (Also on that backlog is a proper ESP8266 port)

Diaoul, 2015-12-04:
Thanks for the very detailed answer. I will reuse the serial and establish
a protocol. It will save me a pin (ESP01 here) and should not be that bad
in terms of performance. I might need to get rid of my DHT22 sensor on the
ESP end due to long delays it requires to measure.

Is Adafruit's lib also disabling interrupts or is this singular to
FastLED's design?

closed 2015-12-04 by Diaoul

focalintent, 2015-12-04:
Yeah the problem is a combination of the ws2811's timing tolerances (on the order of about 3-5 clock cycles) for distinguishing between a 1 and a 0 when writing out data exacerbated by a really short (5us, a mere 80 clock cycles on a 16mhz avr) reset time, so you can't even squeeze much in between pixel data.