FastLED/FastLED #234: Errors compiling Fastled library teensy 3.2

hpvanriemsdijk, 2015-12-08:
Thanks for all the effort you put in maintaining this great library. 

When I try to compile the example "ColorPalette" in Fastled 3.1.0 or 3.1.1 on my teensy 3.1 using arduino IDE 1.6.6 + TeensyDuino 1.26, I get the following errors: 

```
Arduino: 1.6.6 (Windows 10), TD: 1.26, Board:"Teensy 3.2 / 3.1, Serial, 96 MHz optimized (overclock), US English"

In file included from D:\4.Development\1. Arduino\libraries\FastLED\examples\ColorPalette\ColorPalette.ino:1:0:
D:\4.Development\1. Arduino\libraries\FastLED/FastLED.h:12:2: warning: #warning FastLED version 3001000 (Not really a warning, just telling you here.) [-Wcpp]
#warning FastLED version 3001000  (Not really a warning, just telling you here.)
^
D:\4.Development\1. Arduino\libraries\FastLED\examples\ColorPalette\ColorPalette.ino: In function 'void loop()':
ColorPalette:50: error: 'ChangePalettePeriodically' was not declared in this scope
ChangePalettePeriodically();
^
ColorPalette:55: error: 'FillLEDsFromPaletteColors' was not declared in this scope
FillLEDsFromPaletteColors( startIndex);
^
D:\4.Development\1. Arduino\libraries\FastLED\examples\ColorPalette\ColorPalette.ino: In function 'void ChangePalettePeriodically()':
ColorPalette:90: error: 'SetupPurpleAndGreenPalette' was not declared in this scope
if( secondHand == 20)  { SetupPurpleAndGreenPalette();             currentBlending = LINEARBLEND; }
^
ColorPalette:91: error: 'SetupTotallyRandomPalette' was not declared in this scope
if( secondHand == 25)  { SetupTotallyRandomPalette();              currentBlending = LINEARBLEND; }
^
ColorPalette:92: error: 'SetupBlackAndWhiteStripedPalette' was not declared in this scope
if( secondHand == 30)  { SetupBlackAndWhiteStripedPalette();       currentBlending = NOBLEND; }
^
ColorPalette:93: error: 'SetupBlackAndWhiteStripedPalette' was not declared in this scope
if( secondHand == 35)  { SetupBlackAndWhiteStripedPalette();       currentBlending = LINEARBLEND; }
```
On my Arduino boards the library works as expected.

focalintent, 2015-12-08:
Try adding the lines:

```
void ChangePalettePeriodically();
void FillLEDsFromPaletteColors( uint8_t colorIndex);
void SetupPurpleAndGreenPalette();
void SetupBlackAndWhiteStripedPalette();
void SetupTotallyRandomPalette();
```

before setup and see if that fixes things (if it does, it gives me an idea of what needs to be kicked - but it also may be a thing that's gotten broken in teensyduino 1.26 -- the arduino compiler/build system is _supposed_ to effectively do automatically what I described up there).

hpvanriemsdijk, 2015-12-08:
Adding those lines indeed prevent the errors from occurring. 

I tried some other random (non Fastled) sketches in teensyduino 1.26  where the functions are not declared before setup(), so far non of them give the same kind of error.

focalintent, 2015-12-08:
Sometimes the arduino precompiler gets confused and fails to move things up- or it puts them in the wrong place - it has more to do with the structure of the .ino file than anything else.  (In "real world" c/c++ programs - all functions should be declared before they are used, for the arduino compiler, rather than teaching folks about declaration order, the arduino precompiler tries to guess at what function declarations it needs to insert near the top of the .ino file).  

In my setup here, Arduino 1.6.5 w/teensyduino 1.26b1 worked fine with the sketch as it is in the library, Arduino 1.6.5 w/teensyduino 1.26 also worked fine.  Arduino 1.6.6 w/teensyduino 1.26 appears to have an issue - this is most likely an Arduino bug - and not the first that I've heard of in 1.6.6). 

This should be filed against Arduino as a bug in their code - when Genuino Uno is selected, the temporary file that Arduino creates looks like this at the beginning:
```
#include <Arduino.h>
#line 1
#line 1 "/Users/dgarcia/Documents/Arduino/libraries/FastLED/examples/ColorPalette/ColorPalette.ino"
#include <FastLED.h>

#define LED_PIN     5
#define NUM_LEDS    50
#define BRIGHTNESS  64
#define LED_TYPE    WS2811
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100

// This example shows several ways to set up and use 'palettes' of colors
// with FastLED.
//
// These compact palettes provide an easy way to re-colorize your
// animation on the fly, quickly, easily, and with low overhead.
//
// USING palettes is MUCH simpler in practice than in theory, so first just
// run this sketch, and watch the pretty lights as you then read through
// the code.  Although this sketch has eight (or more) different color schemes,
// the entire sketch compiles down to about 6.5K on AVR.
//
// FastLED provides a few pre-configured color palettes, and makes it
// extremely easy to make up your own color schemes with palettes.
//
// Some notes on the more abstract 'theory and practice' of
// FastLED compact palettes are at the bottom of this file.

CRGBPalette16 currentPalette;
TBlendType    currentBlending;

extern CRGBPalette16 myRedWhiteBluePalette;
extern const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM;

void setup();
void loop();
void FillLEDsFromPaletteColors( uint8_t colorIndex);
void ChangePalettePeriodically();
void SetupTotallyRandomPalette();
void SetupBlackAndWhiteStripedPalette();
void SetupPurpleAndGreenPalette();
#line 38
void setup() {
```

But for the teensy - the moved declarations are not there.  They are also not there for the Arduino Zero or Due.

closed 2015-12-08 by focalintent

hpvanriemsdijk, 2015-12-09:
Makes sense, thanks for your help!

focalintent, 2015-12-12:
Here's the arduino bug that's tracking this problem - https://github.com/arduino/arduino-builder/issues/68

netmindz, 2016-04-15:
Any idea what versions of teensyduino or arduino to use/avoid ?

netmindz, 2016-04-15:
1.6.5 looks to work