FastLED/FastLED #235: Arduino Zero Pro

jhozelu, 2015-12-09:
error when compiling a FASTLED 3.1:

Please help me to compile this on this board, I' using Arduino version 1.7.0 they recommended it:

Arduino: 1.7.0 (Windows 8.1), Board: "Arduino Zero Pro (Programming Port)"

Build options changed, rebuilding all

In file included from C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/FastLED.h:38:0,

                 from RGBCalibrate.ino:1:

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h: In member function 'void Pin::_init()':

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h:36:9: error: cannot convert 'volatile PORT_OUT_Type*' to 'RwReg* {aka volatile long unsigned int*}' in assignment

   mPort = portOutputRegister(digitalPinToPort(mPin));

         ^

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h:37:11: error: cannot convert 'volatile PORT_IN_Type*' to 'RoReg* {aka volatile long unsigned int*}' in assignment

   mInPort = portInputRegister(digitalPinToPort(mPin));

           ^

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h: In instantiation of 'static void FastPin<PIN>::_init() [with unsigned char PIN = 9u]':

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h:169:41:   required from 'static void FastPin<PIN>::setOutput() [with unsigned char PIN = 9u]'

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastspi_bitbang.h:37:32:   required from 'void AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::init() [with unsigned char DATA_PIN = 9u; unsigned char CLOCK_PIN = 10u; unsigned char SPI_SPEED = 2u]'

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/chipsets.h:174:3:   required from 'void APA102Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::init() [with unsigned char DATA_PIN = 9u; unsigned char CLOCK_PIN = 10u; EOrder RGB_ORDER = (EOrder)10u; unsigned char SPI_SPEED = 2u]'

RGBCalibrate.ino:73:1:   required from here

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h:161:9: error: cannot convert 'volatile PORT_OUT_Type*' to 'RwReg* {aka volatile long unsigned int*}' in assignment

   sPort = portOutputRegister(digitalPinToPort(PIN));

         ^

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h:162:11: error: cannot convert 'volatile PORT_IN_Type*' to 'RoReg* {aka volatile long unsigned int*}' in assignment

   sInPort = portInputRegister(digitalPinToPort(PIN));

           ^

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h: In instantiation of 'static void FastPin<PIN>::_init() [with unsigned char PIN = 10u]':

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h:169:41:   required from 'static void FastPin<PIN>::setOutput() [with unsigned char PIN = 10u]'

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastspi_bitbang.h:38:33:   required from 'void AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::init() [with unsigned char DATA_PIN = 9u; unsigned char CLOCK_PIN = 10u; unsigned char SPI_SPEED = 2u]'

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/chipsets.h:174:3:   required from 'void APA102Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::init() [with unsigned char DATA_PIN = 9u; unsigned char CLOCK_PIN = 10u; EOrder RGB_ORDER = (EOrder)10u; unsigned char SPI_SPEED = 2u]'

RGBCalibrate.ino:73:1:   required from here

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h:161:9: error: cannot convert 'volatile PORT_OUT_Type*' to 'RwReg* {aka volatile long unsigned int*}' in assignment

   sPort = portOutputRegister(digitalPinToPort(PIN));

         ^

C:\Users\Adminstrador\Documents\Arduino\libraries\FastLED-3.1.0/fastpin.h:162:11: error: cannot convert 'volatile PORT_IN_Type*' to 'RoReg* {aka volatile long unsigned int*}' in assignment

   sInPort = portInputRegister(digitalPinToPort(PIN));

           ^

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

focalintent, 2015-12-09:
I don't yet support the Arduino zero pro - and I'm only supporting up to Arduino 1.6.6 from Arduino.cc - I can look into this more later on but I suspect there's something stupid Arduino.org is doing with their build of Arduino that is preventing the board from being recognized properly by the library during compilation.

jhozelu, 2015-12-09:
The only thing  to say is thank you very much for your  help.your support and dedication to this project it' 's a great contribution to my knowledge

focalintent, 2015-12-09:
I'll try grabbing info on the zero pro tomorrow and see what I can do with the library to make it work with those as well.  (On a good note, the library already supports the processor in there - so I just have to make sure that all the internal bits line up

focalintent, 2015-12-09:
Grab the latest master branch, it should build now!

closed 2015-12-09 by focalintent