FastLED/FastLED #237: ColorTemperature example script 4 inactive LED's after first bright LED

wizbrewery, 2015-12-11:
Hi,  I'm running an adafruit ADA1376 30 LED 1 metre neopixel strip using FASTLED library (Arduino Uno with 5v 4amp supply to neopixel).  The ColorTemperature example is the effect I'm after and for LED's 5 to 30 its working beautifully.  the first one is bright the next 4 are off

I've run both adafruit neopixel library and FastLED ColorPalette and both work as they should.  I have both 1000uF cap across the neopixel input and 500R resistor in line with the short arduino data line.  Thanks

focalintent, 2015-12-11:
This is intentional - and part of how the strip is showing off the effect.  The chosen color temperature is in the first led, then a few leds are left black to provide a gap between the temperature led the color.  So, this line:

```
  fill_rainbow( leds + 5, NUM_LEDS - 5, --starthue, 20);
```

starts filling the rainbow 5 leds into the leds array.  

while these liens:

```
  uint8_t secs = (millis() / 1000) % (DISPLAYTIME * 2);
  if( secs < DISPLAYTIME) {
    FastLED.setTemperature( TEMPERATURE_1 ); // first temperature
    leds[0] = TEMPERATURE_1; // show indicator pixel
  } else {
    FastLED.setTemperature( TEMPERATURE_2 ); // second temperature
    leds[0] = TEMPERATURE_2; // show indicator pixel
  }
``` 

are what set the color temperature.  

If you just want a rainbow across all of your leds you should simply do:

```
void loop() 
{
  static uint8_t starthue = 0;
  fill_rainbow(leds, NUM_LEDS, --starthue, 20);
  FastLED.show();
  FastLED.delay(8);
}
```

closed 2015-12-11 by focalintent

wizbrewery, 2015-12-11:
Thank you Daniel for your quick and informative answer.  will try this
tomorrow