FastLED/FastLED #238: Brightening equivalent of nscale8

Diaoul, 2015-12-13:
I'm looking for the equivalent of `leds[i].nscale8(x)` but to brighten instead of fade. The [wiki](https://github.com/FastLED/FastLED/wiki/Pixel-reference#dimming-and-brightening-colors) shows mostly how to dim colors, not really brighten them.

focalintent, 2015-12-13:
leds[i].maximizeBrightness(x) -- effectively does a scale up, but attempts  to preserve the hue of the colors.  

You can also just increment the brightness of elements by doing something like

leds[i] += 5;

Diaoul, 2015-12-13:
None of those solution do what I need. I want to up scale by a fract8. Here is my workaround:

      CRGB color = leds[i];
      leds[i] += color.nscale8(x);

I wish this could be a one liner and save some RAM by not copying the color.

focalintent, 2015-12-13:
You could just do leds[i] += leds[i].scale8(x).  Also this is not a thing you have to worry about memory usage with - the way you did it would use 3 bytes on the stack - just calling a function can use up to a dozen bytes just saving registers.

Diaoul, 2015-12-13:
It doesn't seem to perform the same as my code, `scale8(x)` doesn't max r, g or b up to 255 it seems. Maybe in order to preserve the proper color?

I really like the `fadeLightBy` and `fadeToBlackBy` functions for scaling down a color, the names are explicit which is always nice in an public API and makes my code much more readable than `nscale8`.
Now from such a complete and powerful library I expected to have the equivalent to brighten a led up to full brightness (255 r, g or b) or maximum fidelity. I know there are other ways around it, I just wish it was included ;)

Thanks

focalintent, 2015-12-13:
So ```leds[i] += leds[i].scale8(x);``` is code equivalent to ```CRGB color = leds[i]; leds[i] += color.nscale8(x);``` - but you're right they won't necessarily go all the way up to 255 - in part because nscale8 will allow values to go down to zero - which means some components of your color might never see increases at all.  Doing ```CRGB color = leds[i]; leds[i] += color.nscale8_video(x);``` might work a bit better, because it will ensure that if your r, g, b values are greater than zero, then no matter the scaling number they will be at least one, so you'll see increases.

I can think about a brightenBy - the trick comes in trying to predict the intent of people who will be using it.  Do they want something that if they continually call it, will eventually start de-saturating and brightening into white?  Or do they want something that keeps the hue and stops brightening when one of r, g, or b is 255 (which is what maximizeBrightness does).  Fading down is easy, you're taking information that's already in the color (the full values of r, g, and b) and cutting stuff out.  Going the other direction is difficult - because you could have, say, a dozen colors that if you scale down by 30 will give you the same RGB values - so, if you try to reverse that, which of said dozen colors do you end up going to?  It's tricky to do well.  (Think of it like lossy jpeg compression - if you crank up the compression in a jpeg really high, you can get a nice small file that still shows what the original file had - but there's nothing you can do to that now compressed jpeg file to get back the full sharpness of the original image, because that information is lost).

(As a side note, this is why I _far_ prefer working in HSV.  Because I can define my color as hue and saturation, and then independently cycle the brightness for fade up/down type effects).

(Also, for another side note - the length of my TODO list for FastLED indicates that it is far far far from complete :)

Diaoul, 2015-12-13:
I totally understand there are many ways to brighten a color and various results. I don't claim to have a solution for that but merely pointing out that it would be helpful.
I'll try to work with HSV, seems easier indeed. Is it possible to still use [those](http://fastled.io/docs/3.1/colorpalettes_8cpp_source.html) predefined palettes?

focalintent, 2015-12-14:
Not yet - I just opened up #239 to remind myself or mark to make HSV versions of the default palettes.  I'll also leave this open - the CRGB data type is about to get a massive overhaul done on it, i'll find a good way to represent this in there when I do that.

tljesse, 2016-03-11:
So I was looking for a solution to the CRGB brightening and perused around the web for ages and stumbled up [this at stackoverflow](http://stackoverflow.com/questions/5560248/programmatically-lighten-or-darken-a-hex-color-or-rgb-and-blend-colors) specifically -- Version 2 RGB --. This is for javascript but I played around a bit and converted it into a C function:

`CRGB shadeRGBColor(int R, int G, int B, double percent) {
  double t=percent<0?0:255;
  double p=percent<0?percent*-1:percent;

  return CRGB((round((t-R)*p)+R), (round((t-G)*p)+G), (round((t-B)*p)+B));
}`

Usage is basically the same as in the stackoverflow example, I just use a percent between -1 and 0 with -1 being black and 0 with the original brightness and color. Hope this is helpful!

kriegsman, 2020-02-17:
Cleaning up old issues; closing this one.

closed 2020-02-17 by kriegsman

cintiqs, 2020-04-27:
none of what i read above works. the only thing that work to scale up brightness is here: https://github.com/FastLED/FastLED/wiki/Pixel-reference

 // Add one CRGB color to another.
  leds[i] += CRGB( 20, 0, 0);

  // Add a constant amount of brightness to all three (RGB) channels.
  leds[i].addToRGB(20);