FastLED/FastLED #240: issue with compiling fastLED examples for ATMega8

korkbaum, 2015-12-14:
I'm using my arduino as ISP to program an ATMega8 (in this case). Compiling and uploading code generally works fine. 

However, with FastLed 3.1 (basically with any example code) I'm getting some SPI related nasty compiler issue saying:
================
D:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega8 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_NG -DARDUINO_ARCH_AVR -ID:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -ID:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard -ID:\Klaus\Documents\Arduino\libraries\FastLED-3.1.0 D:\Temp\build7029299836951456602.tmp\DemoReel100.cpp -o D:\Temp\build7029299836951456602.tmp\DemoReel100.cpp.o 

In file included from DemoReel100.ino:1:0:
D:\Klaus\Documents\Arduino\libraries\FastLED-3.1.0/FastLED.h:12:2: warning: #warning FastLED version 3.001.000 (Not really a warning, just telling you here.) [-Wcpp]
 #warning FastLED version 3.001.000  (Not really a warning, just telling you here.)
  ^
In file included from D:\Klaus\Documents\Arduino\libraries\FastLED-3.1.0/FastLED.h:53:0,
                 from DemoReel100.ino:1:
In file included from D:\Klaus\Documents\Arduino\libraries\FastLED-3.1.0/FastLED.h:53:0,                 from DemoReel100.ino:1:
D:\Klaus\Documents\Arduino\libraries\FastLED-3.1.0/fastspi.h:84:88: error: expected template-name before '<' token
 class SPIOutput<SPI_UART0_DATA, SPI_UART0_CLOCK, SPI_SPEED> : public AVRUSART0SPIOutput<SPI_UART0_DATA, SPI_UART0_CLOCK, SPI_SPEED> {};
                                                                                        ^
D:\Klaus\Documents\Arduino\libraries\FastLED-3.1.0/fastspi.h:84:88: error: expected '{' before '<' token
D:\Klaus\Documents\Arduino\libraries\FastLED-3.1.0/fastspi.h:84:88: error: expected unqualified-id before '<' token
Compiling Issue.
=============================================

Compiling the same code e.g. for ATMega168 works, hence I believe there is some setting wrong or missing for ATMega8. Any help is much appreciated.

Regards
Klaus

focalintent, 2015-12-14:
What package are you using to build for the ATMega8?  (I have nothing installed that gives me an ATMega8 target, since it's not technically a tested/supported platform).

Also - until I fix the hardware SPI support for the ATMega8 you can most likely get things to compile by adding this line to the top of your file (e.g. before the ```#include "FastLED.h```) - 

```
#define FASTLED_FORCE_SOFTWARE_SPI
```

korkbaum, 2015-12-14:
Thanks for having a look at this. I'm using the "Arduino NG or older" option. This allows ATmega168 and ATmega8 as targets.
I've tried to add the suggested force-software-SPI setting. I'm getting the respective warning " #warning "Forcing software SPI - no hardware SPI for you!" (so this was correctly applied) but the final compilation error remains, unfortunately.
Best Regards from Berlin
Klaus

focalintent, 2015-12-14:
Ah - I see the problem - for some reason, while the ATMega8 has a USART that can be run in SPI mode, the header files aren't defining the registers that should be used for it - like UBRR0.  Let me see if I can replace these missing definitions.

focalintent, 2015-12-14:
Gyah - that's because the ATMega8 doesn't have a unified UBRR register.  It's split into two locations that are many bytes apart in memory.

focalintent, 2015-12-14:
Ok - re-pull master, that should let you start building again.

closed 2015-12-14 by focalintent

korkbaum, 2015-12-14:
Daniel, I'm very thankful and impressed how quick you are tackling this. The compilation issue is gone (with and without the software SPI setting) but now I'm getting some content verification problems after uploading (even with the blink sketch) which I don't think I've seen before. I'm struggling to see a connection between your fix and the new issue but maybe you have an idea? 
BR Klaus

focalintent, 2015-12-14:
Unfortunately, I have no ATMega8 with which to try/see this - could you turn on verbose compilation output and verbose upload output - do a build/upload and put a copy of the build log into a gist.github.com and link here, so i can see what's going on?  

(I don't think there's any relation between my fix, which is simply to block compilation of the USART based SPI drivers on the ATMega8, and what you're seeing now)

korkbaum, 2015-12-14:
The upload finally worked - outside Arduino IDE, using WinAVR/avrdude directly.
Again, many thanks for the instant help - even for a pretty old µC like tge mega8.

commit fc0d6b8a (2015-12-14): FIx #240 - don't support USART SPI on the ATMega8 for now, since it's register layout is different enough to not work with some simple text substitutions.