FastLED/FastLED #241: FastLED/chipsets.h ISSUE

awalts, 2015-12-21:
I'm trying to implement the FastLED.chipsets.h code into a project I'm doing, and I keep receiving the following error causing the code to fail compiling. It is:
Arduino: 1.6.7 (Windows 7), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Alex\Documents\Arduino\thundercloud\thundercloud.ino:1:0:

C:\Users\Alex\Documents\Arduino\libraries\FastLED/chipsets.h:266:12: error: missing binary operator before token "("

 #if NO_TIME(500, 1500, 500) 

            ^

C:\Users\Alex\Documents\Arduino\libraries\FastLED/chipsets.h:273:12: error: missing binary operator before token "("

 #if NO_TIME(350, 350, 550) 

            ^

C:\Users\Alex\Documents\Arduino\libraries\FastLED/chipsets.h:280:12: error: missing binary operator before token "("

 #if NO_TIME(400, 400, 450) 

            ^

C:\Users\Alex\Documents\Arduino\libraries\FastLED/chipsets.h:287:12: error: missing binary operator before token "("

 #if NO_TIME(800, 800, 900) 

            ^

C:\Users\Alex\Documents\Arduino\libraries\FastLED/chipsets.h:294:12: error: missing binary operator before token "("

 #if NO_TIME(750, 750, 750) 

            ^

exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Anyone have an idea what's going on?

focalintent, 2015-12-21:
You should be including "FastED.h" - not chipsets.h

awalts, 2015-12-21:
So just that file? I believe chipsets wound up being included since I copied over the entire FastLED folder to my Arduino libraries folder, but that is not what to do, correct? All I need is the the FastLED.h file on it's own?

focalintent, 2015-12-21:
Turn on "verbose output during compilation" and show me a full compile log - also can you upload the code you are trying to build o gist.github.com or pastebin so I can see what you are doing.

focalintent, 2015-12-26:
Checked in a fix for this.  Starting with Arduino 1.6.6 selecting "Sketch | Import Library | FastLED" would add every FastLED header file in a random order.  Put in a bunch of protections for when this happens.

closed 2015-12-26 by focalintent