FastLED/FastLED #242: Phantom FastLED.show()

ottojas, 2015-12-26:
Following code blinks leds without explicitly calling FastLED.show().
Either undocumented feature or bug?? I am a newbie to FastLED.
This code is running on a teensyLC.  It is compiled with arduino 1.6.6

 
[TEMP.txt](https://github.com/FastLED/FastLED/files/72505/TEMP.txt)

focalintent, 2015-12-26:
FastLED.delay calls show - as does clear.

closed 2015-12-26 by focalintent

ottojas, 2015-12-28:
Thanks for info..

Your library is super.  Well designed.  I just recently got involved in
Smart LEDS.  Was delighted to find this.  Sorry to have jumped the gun
on bug/feature report.

regards Otto

focalintent, 2015-12-28:
No worries!  I hope you make wonderful things - feel free to ping here or the g+ group for help (or to show off what you make :)