FastLED/FastLED #243: Slow compilation on Raspi 2 /Platform IO with Mega2560 target

dakman, 2015-12-28:
Platform IO takes about 90 seconds to compile, build and run a very simple sketch implementing FastLED . Has anything changed with speed /compilation time improvements or is this still a known issue? (compling for AVR on RPI)

focalintent, 2015-12-28:
There are some versions of the arduino/avr compiler that are slow.  This is exacerbated if you're using WS2812/NeoPixel leds (as the templated/macro'd assembler used to drive those is fairly complex), and if you are using multiple instances (e.g. using multiple pins) then the compile time will multiply that many times over.  

I would suggest making sure to use the most recent versions of FastLED and try a more recent arduino build (1.6.5 or later) to see if that helps.  Outside of that, the assembler re-writing that would be required to make this any different is unlikely to happen any time in the next few months (if ever - it took a long time to get that code right in the first place - and most people aren't building on something like a pi).

focalintent, 2015-12-28:
See also #154 and #174 for more discussion about this from earlier this year.

closed 2015-12-28 by focalintent

dakman, 2015-12-28:
i noticed that gcc and avr gcc was 4.7 so im upgrading to 4.9 .. hoping this helps.. thanks for your fast response and keep up the great work on FastLED ;) i had many problems with Adafruits neopixel library interfering with serial, servos etc

focalintent, 2015-12-28:
> i had many problems with Adafruits neopixel library interfering with serial, servos etc

I have bad news for you - if you are using neopixel/ws2812 leds, you are going to have similar problems with FastLED.

the problem is the writing out of data for those LEDs requires disabling interrupts for an extended period of time, while libraries like Serial and servos require interrupts to be handled quickly to function properly.  

The solution is to either move to something like a teensy 3.x - where there's enough CPU overhead for me to allow interrupts to be enabled, even while writing out WS2812 data, or to switch to LEDs that don't have the timing requirements that require disabling interrupts, like the APA102 (adafruit DotStars) or LPD8806.

dakman, 2015-12-28:
its actually serial communication i had problems with. servos were fine. but whats strange is when i implemented FastLED and removed the Adafruit_neopixel library serial started working again .... if it works i can live with the slightly slower compilation time (maybe 4.9 will fix this too and ill be golden!)

dakman, 2015-12-28:
so i tested building and uploading FastLED with ino (the older platformio cli tool based on 1.0.1) .. seems to be lighterweight yet less feature rich than platformio but after adding a line break (from an existing clean build) my build time with ino was approx 1 min/60 seconds  ... vs... on platform io (i dont know what its based on.. i think it uses some toolchain indepdent of arduino) but it took approx 1.5 minutes/90 seconds... this is even after upgrading to 4.9..

anyway can i speed this up to help accellerate my dev cycle on the pi vs having to connect the embedded mega to an x86/beefier machine.. eg increase the compiler thread count / process priority on the pi? or what is the root cause of the slowness ? In the past with standard arduino and  adafruit's neopixel libraries the compile time is literally 10-15 seconds from start to finish... yet i dont want to use the library as i mentioned FastLED (after the painful yet necessary slow compilation time) works great and serial works like a charm even with the interrupt issues of neopixel

focalintent, 2015-12-28:
The problem is the complexity of the asm code here (to deal with scaling/dithering/correction) and since it is internal to a single function I doubt there's any thread count games that can be played here.   I _might_ be able to get some improvement if I completely rewrite the asm code there - but even then, it's not ensured.