FastLED/FastLED #244: Unnecessary warning?

blackketter, 2016-01-05:
When building an Arduino project with FastLED, I get the following warning in red:
```
warning: #warning FastLED version 3.001.001 (Not really a warning, just telling you here.) [-Wcpp]
#warning FastLED version 3.001.001  (Not really a warning, just telling you here.)
^
```
I understand that this is important for folks who are working FastLED, but the warning seems unnecessary for regular users of the library.

focalintent, 2016-01-06:
Not at all unnecessary - I want the version number of the library spit out during compilation so that when people are having problems I can get a build log from them and get the exact version of the library that they're compiling with.  I added this in after multiple rounds of helping people out who insisted they had one version of the library when a different version was being used.

This is going to stay in for now.

focalintent, 2016-01-06:
(Once I am sure that no one is using the library on versions of gcc older than 4.4.7 then i'll move this over to a pragma message - but there are some environments that this doesn't work on)

closed 2016-01-06 by focalintent

PerryCodes, 2016-04-02:
Why not at text message summarizing what you just said, rather than "Not really a warning, just telling you here." which left me feeling the same way the OP did until I came across this post???

kelvinmead, 2016-05-04:
i guess that if it didn't say "Not really a warning, just telling you here." then a lot more people would be on the message boards with... my code doesn't work, is it because i got warning #warning FastLED version 3.001.001????

NicoHood, 2016-05-04:
Also IDE 1.6.9 (maybe also older versions) print this:
```
Using library FastLED at version 3.1.1 in folder: /Arduino/sketchbook/libraries/FastLED 
```

distrakt, 2016-11-25:
Hi, chiming in.
1. FastLED is awesome. Seriously, been using it a couple months now and it just rocks.
1. Especially for ESP8266, and disabling interrupts, and and, just awesome.
1. I totally get your support dilemma, "What version are you using." I hear you. Been there lots, as a library author myself.
1. That said, the console clutter of the pragma messages is pretty annoying. Every build, I get output that looks like it might be errors, interfering with my visual parsing for actual errors.
1. Every library author has the same problem, but most dont use pragma messages.
1. Please consider removing the #message pragmas.

Respectfully --> David

```
In file included from LampSketch1.ino:6:0:
/Users/poly/Documents/Arduino/libraries/FastLED-3.1.3/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
 #    pragma message "FastLED version 3.001.003"
                     ^
In file included from /Users/poly/Documents/Arduino/libraries/FastLED-3.1.3/FastLED.h:65:0,
                 from LampSketch1.ino:6:
/Users/poly/Documents/Arduino/libraries/FastLED-3.1.3/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from patterns.h:2:0,
                 from patterns.cpp:1:
/Users/poly/Documents/Arduino/libraries/FastLED-3.1.3/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
 #    pragma message "FastLED version 3.001.003"
                     ^
In file included from /Users/poly/Documents/Arduino/libraries/FastLED-3.1.3/FastLED.h:65:0,
                 from patterns.h:2,
                 from patterns.cpp:1:
/Users/poly/Documents/Arduino/libraries/FastLED-3.1.3/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^

Sketch uses 287,197 bytes (27%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 45,892 bytes (56%) of dynamic memory, leaving 36,028 bytes for local variables. Maximum is 81,920 bytes.
```

Nik1990, 2017-06-01:
This:

[image]

aistisraudys, 2019-06-02:
Try this:

#define FASTLED_INTERNAL
#include <FastLED.h>

JohnnyMa, 2019-06-19:
Thanks @aistisraudys 
It helps me a lot :)