FastLED/FastLED #245: Fourth call to FastLED fails

KeXu001, 2016-01-09:
First and foremost I don't know if this is the right place to post this...if it isn't, so sorry! Otherwise here is the issue I'm having:

My setup consists of an Arduino and a 3-pin LED strip.
All I am doing is alternating the first LED in my strip between several colors, in half-second intervals. Now for some reason, this loop() functions works...
```
void loop() {
  leds[0].setRGB(250,0,0);
  FastLED.show(); delay(500);
  leds[0].setRGB(0,250,0);
  FastLED.show(); delay(500);
  leds[0].setRGB(0,0,250);
  FastLED.show(); delay(500);
}
```
...but this loop() doesn't compile...
```
void loop() {
  leds[0].setRGB(250,0,0);
  FastLED.show(); delay(500);
  leds[0].setRGB(0,250,0);
  FastLED.show(); delay(500);
  leds[0].setRGB(0,0,250);
  FastLED.show(); delay(500);
  leds[0].setRGB(250,250,250); // <--all I did was add another step
  FastLED.show(); delay(500);
}
```

Any reason why this might be happening? It's so simple I don't know where to start. I tried looking through the API and Wiki but didn't come across anything that addressed this.

The colors and order don't matter, just the number of times FastLED.show() is called. Additionally, putting a single FastLED.clear() anywhere in there also causes an error.

If it's helpful, here is what gets printed in the Arduino ide's console area:
```
collect2.exe: error: ld returned 5 exit status

Using library FastLED-3.1.0 at version 3.1.0 in folder: E:\Documents and Settings\NotMyName\My Documents\Arduino\libraries\FastLED-3.1.0 
exit status 1
Error compiling.
```

Thanks in advance

focalintent, 2016-01-09:
Can you put the entire program you're trying to compile in here, and/or also the entire output from the console? (Those messages are usually preceded by more)

Also, what version of Arduino are you using?

KeXu001, 2016-01-09:
Haha I copied the code to a different computer and recompiled...it works!

It was on an Arduino Uno but I think the issue was outside Arduino/FastLED. I was on a really old PC running a janky version of Windows XP...a fifth of the screen's pixels were dead.

Anyway there is nothing wrong with your library! Also thank you for opening your code to people like me. Puttin together some neat stuff

closed 2016-01-09 by KeXu001