FastLED/FastLED #246: FastLED to big

NKPmedia, 2016-01-16:
Hello guys,

I use the library for a little project with the gema of adafruit, but I have a problem with the size of my program.

Is there any way to reduce the size of the library?
Because I only need the Neopixel and HSV color support.

focalintent, 2016-01-16:
The library already strips out anything that you aren't using.  Post your code up on gist.github.com or pastebin and I can take a look at it and see if there's any other places to trim.

NKPmedia, 2016-01-17:
Thank you for your help:
http://pastebin.com/iLtgW8q8

But maybe it would be a better idea to use this library and an own hsv to rgb function :|

focalintent, 2016-01-19:
HSV2RGB isn't really the source of the size - the bulk of it is the completely unrolled assembly output for writing to NeoPixels.  I just got to my hotel room (in the middle of a week and a half of traveling) - I'll try to take a look at a build of your code tonight and see what we can trim out.  (There are some things that can be done that make working with FastLED a little hit more difficult, but will save some program memory - I'll see if I can get that written up for you as well).

focalintent, 2016-05-17:
Check out https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples#managing-your-own-output for how to use controller objects directly  - and the change i made to your code here - https://gist.github.com/focalintent/f7f5217b5efdbc6cab0d6a187c8601a0 - which shaves about 680 bytes off the size of the program (mostly culling out things related to dithering and refresh rate management).

closed 2016-05-17 by focalintent