FastLED/FastLED #247: Error Compiling for ATtiny - "'Serial' was not declared in this scope"

Intechgreater, 2016-01-16:
Getting the output below when attempting to compile for ATtiny85 bare, Adafruit Gema, Trinket 8 & 16Mhz in Arduino IDE 1.6.5, ZIP version. However at least Uno, Flora, Feather, and Pro Trinket work. In case it matters, I did recently uninstall IDE 1.6.7.

[image]

Intechgreater, 2016-01-17:
Is dump() in pixelset.h for debugging? With everything relying on Serial.print() commented out, my sketch compiles.

void dump() const {
    /*
    Serial.print("len: "); Serial.print(len); Serial.print(", dir:"); Serial.print((int)dir);
    Serial.print(", range:"); Serial.print((uint32_t)leds); Serial.print("-"); Serial.print((uint32_t)end_pos);
    Serial.print(", diff:"); Serial.print((int32_t)(end_pos - leds));
    Serial.println("");
    */
}

NicoHood, 2016-01-17:
Probably add
```
void dump() const {
#ifdef SERIAL_PORT_MONITOR
Serial.print("len: "); Serial.print(len); Serial.print(", dir:"); Serial.print((int)dir);
Serial.print(", range:"); Serial.print((uint32_t)leds); Serial.print("-"); Serial.print((uint32_t)end_pos);
Serial.print(", diff:"); Serial.print((int32_t)(end_pos - leds));
Serial.println("");
#endif
}
```

https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/standard/pins_arduino.h#L233

Intechgreater, 2016-01-17:
Thanks NicoHood, that works.

focalintent, 2016-01-18:
Whoops - this shouldn't have gotten committed in the first place - yanked out.

closed 2016-01-18 by focalintent