FastLED/FastLED #248: RFduino + APA102 weirdness...

timfield, 2016-01-19:
Working with RFduino and FastLED 3.1 with both WS2812B and APA102 strips.

Error: if NUM_LEDS for APA102 is between 32 and 63, entire strip is non-functioning. Any other setting for NUM_LEDS (ie. 31 or less or 64+) works great. 

Below is simple Blink code for testing. I tested with both APA102 and WS2812B strips in serial, in parallel and just APA102 by self. The WS2812B strips always blink, no matter NUM_LEDS values. APA102 strip always fails (non-blinking) iff APA102_NUM_LEDS is [32..63]. Identical setup on Uno works fine with all NUM_LEDS values.

#include "FastLED.h"
#define APA102_NUM_LEDS 31
#define WS2812B_NUM_LEDS 48
#define NUM_LEDS APA102_NUM_LEDS+WS2812B_NUM_LEDS
CRGB leds[NUM_LEDS];

void setup() { 
   FastLED.addLeds<APA102, 2, 3, BGR>(leds, APA102_NUM_LEDS);
//   FastLED.addLeds<WS2812B, 4, GRB>(leds, WS2812B_NUM_LEDS);
//   FastLED.addLeds<WS2812B, 4, GRB>(leds+APA102_NUM_LEDS, WS2812B_NUM_LEDS);
}

void loop() { 
  fill_solid(leds, NUM_LEDS, CRGB::Red);
  FastLED.show();
  delay(800);
  fill_solid(leds, NUM_LEDS, CRGB::Black);
  FastLED.show();
  delay(1000);
}

focalintent, 2016-01-19:
Can you add the line:

```
#define FASTLED_FORCE_SOFTWARE_SPI
```

_before_

```
#include "FastLED.h"
```

and let me know if that fixes things or not?  (Let's me know whether the problem is related to the hardware SPI interface or not).  Unfortunately, I am traveling right now and don't have RFDuino hardware with me - but hopefully this is something I can debug without having the hardware here.

timfield, 2016-01-19:
Yes, it DOES work with that added in! Hope that is helpful - and there is no urgency on getting to this for now. And thanks for all you do for this community!!

focalintent, 2016-05-17:
Closing this for now - going to track the hardware SPI related issue in #294

closed 2016-05-17 by focalintent