FastLED/FastLED #250: Support for Wino

marcobaldo, 2016-01-26:
Anybody tried the Wino board? I assumed it would just work as it's using an atmel samd21. The dev board looks identical to an Arduino Zero.

http://www.wino-board.com/index.php/en/wino-board/technical

[image]

focalintent, 2016-01-26:
Given that they haven't shipped anything yet I suspect the answer is no.  

While the board uses the same mcu, their port to pin mapping is completely different.  I'm not going to worry about pin definitions for this board until they ship (and I suspect the adafruit's feather m0+wifi will still end up being the better form factor - the pin layout on this is obnoxious and un-breadboardable).

marcobaldo, 2016-01-26:
They've started shipping to Kickstarter backers - I've received my kits already (the pic above is from me). Is there anything I can do?

focalintent, 2016-01-26:
The Arduino zero uses a SAMD21G18A, while this thing uses a SAMD21E17A -  I'm not sure what the differences are between the two chips, if have to spend some time looking at the data sheet.  

(FastLED does all of its own low level pin access and I/O - which is why it's not enough for a board to be supported in the Arduino ide for it to work - I basically have to spin up definitions for variants as they come up)

focalintent, 2016-01-26:
Ok - it at least builds for the wino - no promises on it working (but then again, i rarely do unless I have physical hardware in hand :)

commit aba957f9 (2016-01-26): Add pin definitions and platform bits for the wino - Fix #250
closed 2016-01-26 by focalintent