FastLED/FastLED #251: "function was not declared in this scope"-error when including FastLED

itsterp, 2016-01-26:
I just upgraded from Arduino 1.6.5-r2 to 1.6.7 and I am now getting errors if I try to call functions that are going to be declared later in the sketch but haven't been declared yet. Example:

    #include <FastLED.h>

    void setup() {
      // put your setup code here, to run once:
    }

    void loop() {
      // put your main code here, to run repeatedly:
    }

    void func1(){
      func2();
    }

    void func2(){
      func1();
    }

If I try to compile this code, I get an error saying:
    sketch_jan26a:14: error: 'func2' was not declared in this scope
    func2();
    ^
    exit status 1
    'func2' was not declared in this scope

If I comment the #include <FastLED.h> line, the code compiles just fine.
I am running the lastest Arduino 1.6.7, latest Teensyduino 1.27, lastest version of FastLED-master and board selected as Teensy 3.2 / 3.1

focalintent, 2016-01-26:
This is a bug in Arduino that will be fixed in 1.6.8 - please see https://github.com/arduino/arduino-builder/issues/68

closed 2016-01-26 by focalintent

itsterp, 2016-01-26:
Great, thanks for the reference!