FastLED/FastLED #254: Arduino Zero Not Working

Pecacheu, 2016-01-31:
I can't get this library to work on the Arduino Zero. My code compiles fine on other boards, but if I try to use the Zero, the compiler gives a bunch of "not declared in this scope" errors for all the functions declared in the code. None of the library's examples will compile either.

To test if the FastLED library was causing this, I made a new file and added an empty function like `void myFunc() {}`. Then I tried calling the function from `setup()` and/or `loop()`, this code compiles fine, of course, but if I add `#include <FastLED.h>`, suddenly it errors out with `myFunc not declared in this scope`!

kasperkamperman, 2016-01-31:
I think Zero is not supported yet, because it uses a difference microprocessor (Intel) and not AVR or ARM like the other Arduino boards.

Pecacheu, 2016-01-31:
I think I saw it on the supported list in the wiki...
Also, it does use an ARM. It's similar to the Due.

focalintent, 2016-01-31:
The zero is supported (and I was just using it the other night for some testing). What you're experiencing may be a bug in the Arduino IDE that should be fixed in 1.6.8.

closed 2016-01-31 by focalintent

kasperkamperman, 2016-01-31:
Sorry, I was confused with the 101.