FastLED/FastLED #256: Extend parallel-output support for ESP

NeoBelerophon, 2016-02-02:
It would be nice if the parallel-output could be extended for esp. 
Running it in over-clocked mode should be fast enough to manage it also in bit bang mode?

focalintent, 2016-02-02:
More likely for the next esp platform than the 8266   (There's a couple other platforms that are in line to get parallel output first).

focalintent, 2016-03-29:
Code complete on this - but it needs to be tested before I push it to the world.  Unclear how well it will work at 80Mhz, it may require 160Mhz to do it's thing.

focalintent, 2016-06-02:
Gyah - the GPIO management and layout of the ESP8266 is a nightmare for getting this to work.  It will likely only work on 4 pins once I'm finished with it.

commit 663c9476 (2016-06-08): Fix #256 - enable 4-way parallel output on the ESP8266 using pins 12,13,14,15 (6,7,5 and 8 on the NodeMCU pin layout/mapping).
closed 2016-06-08 by focalintent