FastLED/FastLED #261: Daisy chaining strips with different color order isn't possible.

gravityfox, 2016-02-17:
I'm using APA102 strips purchased at different times with different densities. 
Unfortunately the older ones use GBR, while the newer ones use BGR.

There doesn't seem to be a way to register two strips on the same pins, with one offset from the other.

There's only one hardware SPI, which i really wanted to use.

On Arduino Mega 2560.

focalintent, 2016-02-17:
There's currently no way to change the rgb ordering part-way through a single controller.  See my comment here on how to manually change the byte ordering for some of the leds before calling show - https://plus.google.com/105715375952928093769/posts/hR7TsmQe72p - specifically the sample code in this gist - https://gist.github.com/focalintent/4ba1367b249f31893d9a 

This is not something that is going to change.

closed 2016-02-17 by focalintent

gravityfox, 2016-02-18:
Alright, I was hoping code like that could be avoided, but thanks anyway. Loving the library.