FastLED/FastLED #262: Serial-based Control for Display Mirroring

DaAwesomeP, 2016-02-18:
I really want to implement [something like this](https://github.com/PaulStoffregen/OctoWS2811/blob/master/examples/VideoDisplay/VideoDisplay.ino). The OcttoWS2811 and Fadecandy libraries have methods to display video or mirror a screen with their high quality dithering. The Fadecandy one even has the Open Pixel Control.

However, they only support WS2811's. I want to use APA104's. Would this be possible and worth the hassle or should I just use WS8211's?

focalintent, 2016-02-18:
It'd be possible - honestly, your biggest problem is going to be getting the data efficiently from the pc to the teensy.  Also - while apa102's have a higher data rate, on paper, I have seen various strips requiring slower data rates as they get longer, which may end up eating any performance benefit to the apa102.  

For video walls/display mirroring, my advice is still generally a fadecandy based setup.

closed 2016-02-18 by focalintent

DaAwesomeP, 2016-02-18:
Thanks for your advice! I guess that I'll just go with the Fadecandy and WS8211B's. If there's no guarentee that there'll be any advantage to the APA102's, then there's no point in spending the extra money. Plus, the Fadecandy has mastered their protocol over USB for maximum bandwidth.