FastLED/FastLED #263: Arduino 101 support

artekw, 2016-02-23:
Hello,

FastLED not work on Arduino 101:
`Arduino:1.6.7 (Linux), Board:"Arduino 101"

In file included from /home/artek/sketchbook/libraries/FastLED/led_sysdefs.h:25:0,
                 from /home/artek/sketchbook/libraries/FastLED/FastLED.h:34,
                 from /home/artek/sketchbook/libraries/FastLED/examples/FirstLight/FirstLight.ino:6:
/home/artek/sketchbook/libraries/FastLED/platforms/avr/led_sysdefs_avr.h:10:20: fatal error: avr/io.h: Nie ma takiego pliku ani katalogu
 #include <avr/io.h>
                    ^
compilation terminated.
exit status 1
`

Do you plan to support that board ?

focalintent, 2016-02-23:
The board just came out a month ago - and uses two CPU architectures that there are currently no FastLED ports for (x86 and ARC).  In addition, it is running under a RTOS environment vs. the OS-less environments that FastLED is most often built for.

So, there is a plan to eventually support the board - but I don't have a guess as to when it would happen.

DrewWeth, 2016-07-17:
BUMP -- What's the status on this?

focalintent, 2016-07-17:
No status yet - there's a backlog of other stuff I need to finish first.

mkeyno, 2016-07-18:
@focalintent  do you any plan to port FastLED on FPGA cpu

focalintent, 2016-07-18:
I'd need something more specific than that to give an answer.  There is the cyprus chips, which are an ARM core w/an FPGA-style ability to extend (though I don't believe it's a full FPGA) - which is on the long term list of platforms to look at.

The immediate work for FastLED includes finishing up the RGBW port/work (and I'm not doing any new platform ports until that is finished because of the amount of low level re-writing that i'm having to do for it).  Then next on the porting list will be the LPC810/812 (already finished for a contract job, needs clean up to pull into the main tree), the K64 (the chip used in the next revision of the teensy 3.x), and then the NRF52 and most likely the ESP32, and then probably the Curie based platforms.  So, short of someone taking out a contract and paying me for it, it will be quite some time before other platforms pop up on the list/schedule.

mkeyno, 2016-07-18:
good luck Danial , I eagerly look for it ,and hope you put ESP32 in front line because it will coming soon (August) with two separated core architecture and can forward all FastLED transaction on one core and left all problem with ESP8266 behind,

Jorgen-VikingGod, 2017-01-04:
I just try to port the FastLED to ESP32 - but there are mass of errors...
The most are depends on the Arduino IDE 1.8 I think - because of changed compiler...

I did already a lot of changes yet - but stuck on this (try to compile simple blink example):
```

In file included from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/examples/Blink/Blink.ino:1:0:
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.005
 #    pragma message "FastLED version 3.001.005"
                     ^
In file included from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastled_esp32.h:5:0,
                 from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms.h:26,
                 from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/FastLED.h:52,
                 from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/examples/Blink/Blink.ino:1:
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::hi()':
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h:23:104: error: 'GP16O' was not declared in this scope
   inline static void hi() __attribute__ ((always_inline)) { if(PIN < 16) { _GPB._GPOS = MASK; } else { GP16O |= MASK; } }
                                                                                                        ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::lo()':
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h:24:104: error: 'GP16O' was not declared in this scope
   inline static void lo() __attribute__ ((always_inline)) { if(PIN < 16) { _GPB._GPOC = MASK; } else { GP16O &= ~MASK; } }
                                                                                                        ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::set(_ESPPIN<PIN, MASK>::port_t)':
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h:25:122: error: 'GP16O' was not declared in this scope
   inline static void set(register port_t val) __attribute__ ((always_inline)) { if(PIN < 16) { _GPB._GPO = val; } else { GP16O = val; }}
                                                                                                                          ^
In file included from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastled_esp32.h:5:0,
                 from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms.h:26,
                 from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/FastLED.h:52,
                 from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/examples/Blink/Blink.ino:1:
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::toggle()':
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h:29:108: error: 'GP16O' was not declared in this scope
   inline static void toggle() __attribute__ ((always_inline)) { if(PIN < 16) { _GPB._GPO ^= MASK; } else { GP16O ^= MASK; } }
                                                                                                            ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static _ESPPIN<PIN, MASK>::port_t _ESPPIN<PIN, MASK>::hival()':
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h:35:87: error: 'GPO' was not declared in this scope
   inline static port_t hival() __attribute__ ((always_inline)) { if (PIN<16) { return GPO | MASK;  } else { return GP16O | MASK; } }
                                                                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h:35:116: error: 'GP16O' was not declared in this scope
   inline static port_t hival() __attribute__ ((always_inline)) { if (PIN<16) { return GPO | MASK;  } else { return GP16O | MASK; } }
                                                                                                                    ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h: In static member function 'static _ESPPIN<PIN, MASK>::port_t _ESPPIN<PIN, MASK>::loval()':
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h:36:87: error: 'GPO' was not declared in this scope
   inline static port_t loval() __attribute__ ((always_inline)) { if (PIN<16) { return GPO & ~MASK; } else { return GP16O & ~MASK; } }
                                                                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/platforms/esp/32/fastpin_esp32.h:36:116: error: 'GP16O' was not declared in this scope
[... cut ...]
                 from /Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/examples/Blink/Blink.ino:1:
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/fastspi.h: At global scope:
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
exit status 1
Error compiling for board ESP32 Dev Module.
```

focalintent, 2017-01-04:
These errors are all because you are still using ESP8266 GPIO register names/descriptions/layouts.  You need to re-write fastpin_esp32.h to use the ESP32 GPIO registers.

Theamazingarnaud, 2017-01-13:
Hello, did anyone have any luck with running Fastled on the esp32 ?
this is the error I get, I know I should be more patient but this board is awesome :)

`rom /Users/aatchi99/Documents/Arduino/libraries/FastLED/led_sysdefs.h:27:0,
                 from /Users/aatchi99/Documents/Arduino/libraries/FastLED/FastLED.h:44,
                 from /Users/aatchi99/Documents/Arduino/libraries/FastLED/examples/DemoReel100/DemoReel100.ino:1:
/Users/aatchi99/Documents/Arduino/libraries/FastLED/platforms/avr/led_sysdefs_avr.h:12:20: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
                    ^
compilation terminated.
Multiple libraries were found for "FastLED.h"
 Used: /Users/aatchi99/Documents/Arduino/libraries/FastLED
 Not used: /Users/aatchi99/Documents/Arduino/libraries/arduino_760590
exit status 1
Error compiling for board ESP32 Dev Module.
`

Jorgen-VikingGod, 2017-01-15:
@Theamazingarnaud I'm on the last mile to get it working...
Hopefully I have time today to work on it; cause I'm really busy (work and family)

Theamazingarnaud, 2017-01-16:
@Jorgen-VikingGod , No worry, family always first :)

Jorgen-VikingGod, 2017-01-17:
Sorry guys - I tried to look again on this part of code; but this is too heavy to figure out by me. I'm not that low-level guy :/
As I can see, the ESP32 stuff has a completely changed code style compared to the ESP8266...
So it is not just, change GPIO names - the complete register of the pins was changed...

Hope someone with more knowledge can do this port. I currently do not have the time to do this - sorry for that.

eshkrab, 2017-03-13:
Hello!
Is anyone else attempting esp32 ports? I started but I also don't know enough low-level programming to be very effective. Here's a link to my repo if anyone wants, currently stuck at trying to figure out the correct way of writing to registers for esp32:
https://github.com/eshkrab/FastLED-esp32

Jorgen-VikingGod, 2017-03-20:
@eshkrab maybe you can peek into the PullRequest of [Adafruit_NeoPixel](https://github.com/adafruit/Adafruit_NeoPixel/pull/125) which brings support for ESP32.

wjgoodrich, 2017-06-20:
I would also be very interested in a port for the ESP32.  @eshkrab what's the current status of your attempt?  Thanks.

HorstBaerbel, 2017-08-02:
+1 Would be sweet being able to use the ESP32. Now that [all Intel Arduino boards including Curie are dead](http://hackaday.com/2017/07/25/the-end-of-arduino-101-intel-leaves-maker-market/) there's no use porting FastLED to those anyway, freeing up time for ESP32... ;)