FastLED/FastLED #266: Arduino/ESP8266 2.1.0 now includes __cxa_guard_* functions

ncaunt, 2016-02-29:
As of version 2.1.0, ESP8266 core for Arduino includes the __cxa_guard_* functions which conflict with those in FastLED.

I haven't yet had a chance to find a neat solution but for the time being removing the line `#define NEED_CXX_BITS` from platforms\esp\8266\led_sysdefs_esp8266.h fixes the issue.

Sample messages to help those searching for the error:

```C:\Users\foobar\AppData\Local\Temp\buildd782c7fbc1e7873555d9cadaa1f56454.tmp/arduino.ar(abi.cpp.o): In function `__cxa_guard_acquire':

C:\Users\foobar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/abi.cpp:67: multiple definition of `__cxa_guard_acquire'

C:\Users\foobar\AppData\Local\Temp\buildd782c7fbc1e7873555d9cadaa1f56454.tmp\libraries\FastLED\FastLED.cpp.o:C:\Program Files (x86)\Arduino\libraries\FastLED/FastLED.cpp:252: first defined here

C:\Users\foobar\AppData\Local\Temp\buildd782c7fbc1e7873555d9cadaa1f56454.tmp/arduino.ar(abi.cpp.o): In function `__cxa_guard_release':

C:\Users\foobar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/abi.cpp:78: multiple definition of `__cxa_guard_release'

C:\Users\foobar\AppData\Local\Temp\buildd782c7fbc1e7873555d9cadaa1f56454.tmp\libraries\FastLED\FastLED.cpp.o:C:\Program Files (x86)\Arduino\libraries\FastLED/FastLED.cpp:257: first defined here

C:\Users\foobar\AppData\Local\Temp\buildd782c7fbc1e7873555d9cadaa1f56454.tmp/arduino.ar(abi.cpp.o): In function `__cxa_guard_abort':

C:\Users\foobar\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/abi.cpp:84: multiple definition of `__cxa_guard_abort'

C:\Users\foobar\AppData\Local\Temp\buildd782c7fbc1e7873555d9cadaa1f56454.tmp\libraries\FastLED\FastLED.cpp.o:C:\Program Files (x86)\Arduino\libraries\FastLED/FastLED.cpp:261: first defined here

collect2.exe: error: ld returned 1 exit status

Using library FastLED at version 3.1.1 in folder: C:\Program Files (x86)\Arduino\libraries\FastLED 
exit status 1
Error compiling.```

TheAustrian, 2016-03-01:
I had the same issue - thank you for the quick fix!

timfield, 2016-03-05:
Same issue here too... this temporary workaround worked. Thanks!

antonmeyer, 2016-03-06:
proposal:
cxxabi.h
sets #ifndef _CXXABI_H
#define _CXXABI_H 1

so you might check for this

focalintent, 2016-03-08:
cxxabi.h is included in the older versions of the esp8266 library code, and _CXXABI_H is defined, so that wouldn't help for this.

NEED_CXX_BITS is necessary for some platforms, and not for others, and I haven't found a reliable way to make the determination of whether it is or isn't, which is why it's defined by hand for various platforms.

I'll have to work out another way to fix this that will work with esp8266 environments that do and don't provide these definitions.

focalintent, 2016-03-08:
2.1.0 isn't yet available in the boards manager from the stable release url - makes testing fixes for this irritating :)

closed 2016-03-08 by focalintent

antonmeyer, 2016-03-12:
works. I did not know about this tricks.

tomjuggler, 2016-08-02:
Thank you ncaunt I am saved by your trick. Circuit boards arriving this afternoon and I need to program my esp's like now!

tomjuggler, 2016-08-02:
Oh I see updated FastLED it is fixed this issue is closed anyway duh. Thanks guys for fixing it I had a minor panic attack there

commit 55cf5f79 (2016-03-08): Fix #266 - make the cxxbits weak so they can be overriden if present