FastLED/FastLED #268: teensy-lc ws2812, first led in color order flickers

pi-ratte, 2016-03-03:
Hi

When using my teensy-lc with ws2812( might be APA104)  the first led in the color order flickers.
when using apa102 i dont have this problem.

if i change the color order the flicker moves to that color.

After pin 17 the 5V pin of the Teensy-lc gave me this problem i hooked up a 74htc245N to pin 15 and i have the same results.

I think there actually APA104 leds, if set to APA104 i have the same problem.

if i switch from FastLed.delay() to delay the leds just dont flicker as much.

here is the minimalistic code i am using.

thanks for the Help :)

`
//#define FASTLED_FORCE_SOFTWARE_SPI
#include <FastLED.h>

#define LED_PIN     15
//#define CLK_PIN     13
#define NUM_LEDS    3
#define BRIGHTNESS  255
#define LED_TYPE    APA104
#define COLOR_ORDER BRG
CRGB leds[NUM_LEDS];
#define UPDATES_PER_SECOND 3

void setup() {
    delay( 3000 ); // power-up safety delay
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
    FastLED.setBrightness(  BRIGHTNESS );
    FastLED.setDither( 0 );
     Serial2.begin(115200);
     Serial2.println("hello");
}
void loop()
{
   
    leds[0] = (CRGB::Red);
    leds[1] = (CRGB::Green);
    leds[2] = (CRGB::Blue);
 
    FastLED.show();
      FastLED.delay(1000 / UPDATES_PER_SECOND);
 
}

`

marmilicious, 2016-03-03:
FastLED.delay() calls FastLED.show() under the hood, so if you have:

FastLED.show();
FastLED.delay(1000 / UPDATES_PER_SECOND);
you will actually be calling show() twice in a row which isn't needed.

Instead of:
FastLED.addLeds(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
what if you try:
FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS);
so you are specifically setting everything as you have defined above.  If you just use FastLED.addLeds(leds, NUM_LEDS) this might work, but maybe it's defaulting to something unexpected.  Perhaps test with specifying everything first before using an abbreviated version of addLeds.

Also, do you actually have a chip that is hTC? or is it hCT?

pi-ratte, 2016-03-03:
Hi thanks for the quick reply.
about the chip sorry its a  HCT of course.  and i tested with  a  330 and 100 Ohm resistor( forgot about that in the post)

ok changes made : 
1 : removed FastLed.show() since we are calling the FasLed.Delay
makes no difference same effect. led[1] flickers, if i change the collor order it just flickers in a different color. 

2: change to ` FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS); ` 
now led[0] stays off, led[1] flickers.

marmilicious, 2016-03-03:
That's strange, not sure about leds[0] doing that now.
Did you try the RGBCalibrate example to confirm your color order?  Even if there's still flickering, at least it could be good to know that for sure.

Just to confirm/try to get as much info as possible about the setup:
What Arduino IDE version are you using?
And you have the latest version of the FastLED library?
What OS are you using?
How are you powering the led strip? 
You have a resistor on data line?
Are you still using the 74HCT245N in the current setup?

pi-ratte, 2016-03-03:
Hi again. The color order changing is just to show that the problem stays the same if i change it so its allways the first led in the LED chip, as if  the communication for the first led is messed up.  
the actual collor order is GRB. i have been using these strips without problems on my Mega2560, ATTiny85 and  UNO.

it just when using the Teensy-lc that it messes up.

What Arduino IDE version are you using:  1.6.7
And you have the latest version of the FastLED library Github from last week. Library manager says 3.1.1
What OS are you using: windows 10
How are you powering the led strip?: does not matter how. im just taking it from the tennsy-lc but i have added a psu no change.
You have a resistor on data line? yes also tried without does not change anything  [https://www.pjrc.com/teensy/td_libs_OctoWS2811.html](as described here)
Are you still using the 74HCT245N in the current setup? yes i also tried directly to pin 17 since it is 5V.

APA102 leds work like a charm it just the WS2812 type that is causing problems.

pi-ratte, 2016-03-03:
rence

pi-ratte, 2016-03-03:
sorry github messing with me...  not fixed

focalintent, 2016-03-08:
I will have to wait until I get a working teensy-lc to see what the output looks like, it's possible a recent gcc version change is messing with the asm output.

focalintent, 2016-03-25:
Also - try grabbing the most recent - I fixed a different m0 related issue just now - possible that it fixed your problem.

pi-ratte, 2016-03-27:
Just did a test with the latest github and it looks great :) Thanks for the fix !

closed 2016-03-27 by pi-ratte

00fractal, 2016-07-26:
Hi,  I think  I've found the  thing you were looking for, just take a  look here  <http://private.snehankur.org/e4usex>

My best to you, richard

7577