FastLED/FastLED #269: Green does not work on WS2812B

talolard, 2016-03-05:
Hello, 
similar to #104 
I'm using a Mega2650 with a WS2812B strip. 
Running the RGB test example, I found my strip is BRG.
Red and Blue appear fine, but the green is completely off for any value between 0-255. 
Any ideas?
Thanks

davemaster, 2016-03-06:
check cable/connection of green led-component...

focalintent, 2016-03-08:
What pin are you using (there used to be problems with some pins on the mega - though those were fixed a while ago - and the problem caused didn't manifest this way). 

This is not similar to #104 - that was caused by timing issues with the first bit being written out causing green to always be on.

What version of the library are you using - have you tried grabbing the most recent version of the library out of github, and can you please post your code to gist so I can see what you're trying to run.

@davemaster given that these chips are WS2812B's, there is no "cable/connection of green led-component" - the led has 4 pins, Vin, Gnd, Din, Dout - the red, blue, green leds are part of the chip.

talolard, 2016-03-10:
Hi.
I'm using pin 8.
I'm using version 3.1.
The code I'm running is the [RGBCalibrate example](https://github.com/FastLED/FastLED/blob/master/examples/RGBCalibrate/RGBCalibrate.ino)
[gist that works] (https://gist.github.com/talolard/bc4af5aece5c3ac3c2e6).
I just noticed that running with FastLED I used the WS2812B, and in this example I used Adafruits default NEOPIXEL.

focalintent, 2016-03-10:
Try this updated version - https://github.com/FastLED/FastLED/blob/master/examples/RGBCalibrate/RGBCalibrate.ino (using WS2812B)

Also - try switching to pin 5 - the mega's have had problems with some of the pins in the past.

focalintent, 2016-03-25:
Ok - make sure that you've pulled the most recent version of FastLED from github - I've just gotten a setup working here and was able to test my Mega2560 with WS2812's on pin 8 and and everything is working fine.

Also, what version of arduino are you using?  (Please update to the most recent version).

Since I have it working here, and I haven't heard any updates from you in a couple of weeks, i'm going to close this for now, feel free to re-open it if you're still having problems.

closed 2016-03-25 by focalintent