FastLED/FastLED #270: CRGB leds[700]; Memory footprint too large for Arduino nano/uno

ThibHub, 2016-03-07:
Hello,
Your library is so fast that I can use it on extremely long stripes! 
Unfortunately I am forced to use an Arduino Mega which brings a form factor/cost problem.

Because I use less than 8 fixed colors, is there a way to allocate a 8 colors palette (or 256 colors palette) and change library files to get a source code something like this : 

char Leds[700];//allocates only 700 bytes instead of 3x700 bytes
CRGB PalletMatch[8]={CRGB::Black; CRGB::Red; CRGB::Orange; CRGB::Green; CRGB::Blue; CRGB::Pink; CRGB::Yellow; CRGB::White;}//something to define corresponding colors of course 1 byte allows 256 colors
...
FastLED8.addLeds<WS2812B, DATA_PIN>(Leds, 700);//new Setup()
...
FastLED8.Show() // new show() method

The FastLED8.Show() would get the byte color from Leds[] and then lookup into the PalletMatch to throw the corresponding RGB values to the WS2812B... even hardcoded colors without PalletMatch would work for me.

If no one sees interest in adding this new feature to FastLed.h I'd like to have a look into the library code but do not know where to start...

thanks for guidance,
Thibaut

TheAustrian, 2016-03-07:
Is it out of the question to just use another small and cheap Arduino-alternative like one of the small ESP8266 boards? Those should be small enough and they are very cheap at 3-5 dollars and have tons more storage and processing capacity.

kriegsman, 2016-03-07:
Someday we do hope to have a palette/indexed-color mode... and we've done a lot of planning and thinking about it, but it isn't going to be anytime very soon, just because we both have full-time jobs at a startup etc. 

In the meantime, I think we'd recommend what TheAustrian said: try an ESP8266 or a Teensy 3.x!

ThibHub, 2016-03-07:
okay, I will have a look to SRAM specs of both ESP8266 and Teensy 3.x
Keep the excellent job, I am impressed with FastLED I do not see RT clock deviation (compared to PololuLedStrip library even using interupt_friendly mode) although I throw kilos of bytes to very long stripes... Well done!
Take care.

closed 2016-05-17 by focalintent