FastLED/FastLED #271: ESP8266: FastLED + MQTT = stack strace

hggh, 2016-03-12:
Platform: ESP8266, NodeMCU 1.0 (ESP12E)
FastLED Version: master branch from 10. March
PubSubClient: Version 2.6

If using the MQTT example for ESP8266 from https://github.com/knolleary/pubsubclient/blob/master/examples/mqtt_esp8266/mqtt_esp8266.ino is works.

if I add only the fastLED stuff and do one loop in the setup() the first client.connected() in the loop() crashes:

<code>
Exception (9):
epc1=0x402044d5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3f0000ff depc=0x00000000

ctx: cont 
sp: 3ffefdd0 end: 3ffeffb0 offset: 01a0

3ffeff70:  3ffeed40 feefeffe feefeffe feefeffe  
3ffeff80:  feefeffe 3ffeed40 3ffeef7c 4020271d  
3ffeff90:  3fffdad0 00000000 3ffeef7c 40203f48  
3ffeffa0:  feefeffe feefeffe 3ffeef90 40100958  

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
</code>

the stack strace with the decoder:

<code>
0x402044d5: PubSubClient::connected() at Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 586

0x4020271d: loop at Arduino/libraries/FastLED/controller.h line 170

0x40203f48: loop_wrapper .arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_main.cpp line 43

0x40100958: cont_norm at .arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/cont.S line 109
</code>

The sample code is available here, it will crash: https://gist.github.com/hggh/a578505703f34056589f

Perhaps the problem is between FastLED and pubsubclient
 (@knolleary)

hggh, 2016-03-12:
you don't need to connect the LEDs to your esp, it will crash anyway.

focalintent, 2016-03-13:
wdt reset is the watchdog timer, that loop you have in setup runs for 4.5 seconds - with the esp I do t think you can block the word for that long, certainly for not coming out of setup.

PhilG, 2016-04-05:
Since I am interested in maybe using FastLED and MQTT on my ESP as well, could it work if his loop would not be in setup() but elsewhere? Afaik delay work similar to yield and gives the ESP time to do his background legwork of keeping up WiFi etc...

focalintent, 2016-04-05:
I suspect so - the problem with doing this in setup is, being setup, it's possible there's other setup work that won't happen until after setup returns.  That crash info indicates that something is triggering the watchdog timer.  There are folks on the G+ group who I believe are using mQTT without any problems.

PhilG, 2016-04-06:
I am not getting a stacktrace with this code simply delaying the return of `setup()`

```arduino
void setup() {
  Serial.begin(115200);
  Serial.println ("Starting delay of 6 seconds!");

  for(int i = 0; i < 200; i++) {
    delay(30);
  }
  
  Serial.println("Delay didn't bother setup()");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Seems like it is something else or a combination I guess

TheAustrian, 2016-04-06:
Just try putting the for-loop into the loop() function and make it run only once with a conditional (for example initialize a bool variable as true, check in an if-statement, then set it to false - that way the code only gets executed once).

The ESP is a bit picky about taking too much time in the setup() function iirc.

focalintent, 2016-04-06:
Delay isn't going to block interrupts the way that writing out led data will.

focalintent, 2016-04-06:
(Again, the wdt in the original stack trace means that something spun in a tight loop long enough that the watchdog timer was never reset, forcing a crash/reboot)

closed 2016-05-17 by focalintent