FastLED/FastLED #273: not declared in this scope

phil71994, 2016-03-14:
Hello, i hope somebody can help me on this one.

i am not much of a programmer so my problem may seem easy to fix for most of you. 

i loaded up the fire2012 template and tried to flash it, but everytime i get this error message:

Arduino: 1.6.7 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

> G:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "G:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Philipp\AppData\Local\Arduino15\packages" -tools "G:\Program Files (x86)\Arduino\tools-builder" -tools "G:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Philipp\AppData\Local\Arduino15\packages" -built-in-libraries "G:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Philipp\Documents\Arduino\libraries" -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10607 -build-path "C:\Users\Philipp\AppData\Local\Temp\build950321a8f98f8942bd43f1b65bc0625c.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Philipp\AppData\Local\Temp\arduino_950321a8f98f8942bd43f1b65bc0625c\Fire2012.ino"
> G:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "G:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Philipp\AppData\Local\Arduino15\packages" -tools "G:\Program Files (x86)\Arduino\tools-builder" -tools "G:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Philipp\AppData\Local\Arduino15\packages" -built-in-libraries "G:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Philipp\Documents\Arduino\libraries" -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10607 -build-path "C:\Users\Philipp\AppData\Local\Temp\build950321a8f98f8942bd43f1b65bc0625c.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Philipp\AppData\Local\Temp\arduino_950321a8f98f8942bd43f1b65bc0625c\Fire2012.ino"
> "C:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "C:\Users\Philipp\AppData\Local\Temp\build950321a8f98f8942bd43f1b65bc0625c.tmp\sketch\Fire2012.ino.cpp" -o "nul"
> "C:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\Philipp\Documents\Arduino\libraries\FastLED" "C:\Users\Philipp\AppData\Local\Temp\build950321a8f98f8942bd43f1b65bc0625c.tmp\sketch\Fire2012.ino.cpp" -o "nul"
> "C:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\Philipp\Documents\Arduino\libraries\FastLED" "C:\Users\Philipp\Documents\Arduino\libraries\FastLED\FastLED.cpp" -o "nul"
> "C:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\Philipp\Documents\Arduino\libraries\FastLED" "C:\Users\Philipp\Documents\Arduino\libraries\FastLED\colorpalettes.cpp" -o "nul"
> "C:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\Philipp\Documents\Arduino\libraries\FastLED" "C:\Users\Philipp\Documents\Arduino\libraries\FastLED\colorutils.cpp" -o "nul"
> "C:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\Philipp\Documents\Arduino\libraries\FastLED" "C:\Users\Philipp\Documents\Arduino\libraries\FastLED\hsv2rgb.cpp" -o "nul"
> "C:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\Philipp\Documents\Arduino\libraries\FastLED" "C:\Users\Philipp\Documents\Arduino\libraries\FastLED\lib8tion.cpp" -o "nul"
> "C:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Philipp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rt [... cut ...]
> C:\Users\Philipp\AppData\Local\Temp\arduino_950321a8f98f8942bd43f1b65bc0625c\Fire2012.ino: In function 'void loop()':
> 
> Fire2012:26: error: 'Fire2012' was not declared in this scope
> 
>    Fire2012(); // run simulation frame
> 
>             ^
> 
> Bibliothek FastLED in Version 3.1.1 im Ordner: C:\Users\Philipp\Documents\Arduino\libraries\FastLED  wird verwendet
> exit status 1
> 'Fire2012' was not declared in this scope

i get these kind of errors with nearly all the templates in the fast led library

TheAustrian, 2016-03-14:
Move all functions like Fire2012() before the loop - then it should work. I had to do that with a lot of examples, too...

focalintent, 2016-03-14:
This is an ongoing bug with the Arduino IDE - many of these issues have been fixed in 1.6.8.

closed 2016-03-14 by focalintent

phil71994, 2016-03-14:
@TheAustrian thank you, it works now :)

admintei28, 2019-02-12:
/*----------------------------------------------------------------------
 * Project    : Jam Digital DotMatrix Display Detik Geser
 * Author     : Amar Lakar Cengkal
 * Version    : 1.0
 * 
 ----------------------------------------------------------------------*/

#include <DMD2.h>         
#include <fonts/SystemFont5x7.h> 
#include <fonts/MyBigFont.h> 
#include <fonts/Arial_Black_16.h> 
#include <stdio.h>
#include <DS1302.h>

#define Panjang 1                        
#define Lebar 1 
#define max_char 50                        

SoftDMD dmd(Panjang, Lebar);                
char message[max_char];
char tahun[5];
char bulan[7];
char tanggal[7];
char jam[7];
char menit[7];
char detik[7];
char hari[7];
char tulisan[20];

char r_char;
byte asu= 0;
int i,r;
int a=0;
int taek=0;

namespace {

const int kCePin   = A2;  // Chip Enable
const int kIoPin   = A1;  // Input/Output
const int kSclkPin = A0;  // Serial Clock

DS1302 rtc(kCePin, kIoPin, kSclkPin);

String dayAsString(const Time::Day day) {
  switch (day) {
    case Time::kSunday: return "Minggu";
    case Time::kMonday: return "Senin";
    case Time::kTuesday: return "Selasa";
    case Time::kWednesday: return "Rabu";
    case Time::kThursday: return "Kamis";
    case Time::kFriday: return "Jumat";
    case Time::kSaturday: return "Sabtu";
  }
  return "(unknown day)";
}
/////////////////////////////////////////////////////////////////////
void setjam()
{
     if(Serial.available())
  {
    for(i=0; i<50; i++)
    {
    message[i] = '\0';
    }
    for(i=0; i<7; i++)
    {
      tahun[i] = '\0';
      bulan[i] = '\0';
      tanggal[i] = '\0';
      jam[i] = '\0';
[... 550 lines cut ...]
   
   dmd.clearScreen();
   dmd.drawString( k, 0, "KAM");
   delay(55);
   
  }

setjam();
if(tulisan[0]!='x')
 {
  dmd.clearScreen();
  dmd.selectFont(Arial_Black_16);
  for(int k=33;k>=-240;k--)
  {  
   dmd.clearScreen();
   dmd.drawString( k, 0, tulisan);
   delay(55);   
  }
 }
   a=0;
  }
   
}

'MyBigFont' was not declared in this scope