FastLED/FastLED #275: WIP: Porting FastLED to CC3200

drewrisinger, 2016-03-24:
Hi, I'm currently working on porting the FastLED library to the TI CC3200, for part of my undergraduate EE senior project. I'm really impressed with the level of support available for the CC3200 and it's wide range of networking functions. It's like a bigger, more flexible ESP8266, and it even includes cool features like OTA updates.

If anyone has any experience working with the TI compiler (or a lot of these low-level C macros similar to Teensy LC/KL26 support is done), I would love the assistance!

Full repo & code can be found at my fork: https://github.com/drewrisinger/FastLED-CC3200

Right now I'm mostly having troubles working through the following compiler errors dealing with templates & specialization. Any help would be appreciated
```
"C:\Users\[...]\FastLED\platforms\arm\cc3200\fastpin_arm_cc3200.h", line 76: error #793: 
          explicit specialization is not allowed in the current scope
  template<> class FastPin<0> : public _ARMPIN<0, (1 << 0), _R(L, GPIO_O_GPIO_DATA), _R(L, GPIO_O_GPIO_DIR)> {};
  ^

"C:\Users\[...]\FastLED\platforms\arm\cc3200\fastpin_arm_cc3200.h", line 76: error #505: 
          class "FastPin<(uint8_t)'\000'>" cannot be specialized in the
          current scope
  template<> class FastPin<0> : public _ARMPIN<0, (1 << 0), _R(L, GPIO_O_GPIO_DATA), _R(L, GPIO_O_GPIO_DIR)> {};
```

focalintent, 2016-03-24:
You haven't checked in fastpin_arm_cc3200.h - so I can't see what it is you're trying to do.  My guess is you don't quite have the gpio registers set up/defined correctly.  

(and wow, looking at the datasheet for the cc3200 - they actually managed to come up with a sloppier gpio register setup than STM!  I'm impressed...)

drewrisinger, 2016-03-26:
I suppose that it would help if I checked in the actual file, but the lines that were causing errors are included in above file. I got their GPIO register setup worked out (at least for just setting/clearing bits), using the register assignment that TI uses in their Hardware Abstraction Layer.

For the templates, I'm basically using the same model that was used in the Teensy 3/LC ports, using macro expansions to expand the _DEFPIN_ARM macro and extend/specialize the FastPin class.

I'm still getting the following error (after switching to the G++ compiler from TI's compiler), using the following two lines of code (not in the [file I checked in](https://github.com/drewrisinger/FastLED-CC3200/blob/master/platforms/arm/cc3200/fastpin_arm_cc3200.h), I just manually expanded the _DEFPIN_ARM macro to eliminate that as a source of error)

```
template<>
class FastPin<0> : public _ARMPIN<0, (1 << 0), _R(L, GPIO_O_GPIO_DATA), _R(L, GPIO_O_GPIO_DIR)> {};
```

Compiler error (G++, std=c++11)
```
C:/Users/risingerar1/Documents/GitHub/FastLED/platforms/arm/cc3200/fastpin_arm_cc3200.h:76:10: error: explicit specialization in non-namespace scope 'struct __gen_struct_A0_GPIO_O_GPIO_DATA::__gen_struct_A0_GPIO_O_GPIO_DIR::__gen_struct_A1_GPIO_O_GPIO_DATA::__gen_struct_A1_GPIO_O_GPIO_DIR::__gen_struct_A2_GPIO_O_GPIO_DATA::__gen_struct_A2_GPIO_O_GPIO_DIR::__gen_struct_A3_GPIO_O_GPIO_DATA::__gen_struct_A3_GPIO_O_GPIO_DIR'
 template<>
          ^
C:/Users/risingerar1/Documents/GitHub/FastLED/platforms/arm/cc3200/fastpin_arm_cc3200.h:77:7: error: specialization of 'template<unsigned char PIN> class FastPin' must appear at namespace scope
 class FastPin<0> : public _ARMPIN<0, (1 << 0), _R(L, GPIO_O_GPIO_DATA), _R(L, GPIO_O_GPIO_DIR)> {};
```