FastLED/FastLED #278: Arduino Zero / MKR1000 : first byte of clockless pixel data not sent

Daniel-dk, 2016-03-25:
This *might* be related to issue #269 ( missing byte of data )
TLDR: it doesn't look like the first byte is being sent with clockless controller
using WS2812B LEDs with Arduino Zero / MKR1000 ( ATSAMD21G18A )
- the pin/port mapping is different but does the same on both controllers/bootloaders 
- I added the MKR1000 pin map to fastpin_arm_d21.h
- Data doesn't seem to go missing when I use WS2801 ( SPI like ) controller, only for the clockless WS2812B .
- Timing also seems a bit strange but I am unsure where to change it only for this platform - tweaking in \FastLED\platforms\arm\common\m0clockless.h ? but Assembler is voodoo to me

attached are Logic analyser (saleae logic) captures of the following code:

```
// MKR1000
#define DATA_PIN 15 // 14 on Arduino Zero
#define LOGIC_PIN 6 // 6 on Arduino Zero
// How many leds in your strip?
#define NUM_LEDS 1
// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() {
        // - keep RGB mapping as "RGB" so we know which byte is missing
	FastLED.addLeds<WS2812B, DATA_PIN,RGB>(leds, NUM_LEDS); // WS2812B pixel
	FastLED.addLeds<WS2812B, LOGIC_PIN, RGB>(leds, NUM_LEDS); // Logic analyser

// - so Fastled doesn't correct our output
	FastLED.setBrightness(255);
	FastLED.setCorrection(UncorrectedColor);
	FastLED.setDither(DISABLE_DITHER);
}

void loop() {
	leds[0] = CRGB(255,255,255); // White
	FastLED.show();
	leds[0] = CRGB(255, 0, 0);  // red/green depending on order
	FastLED.show();
	leds[0] = CRGB(0, 255, 0);  // red/green depending on order
	FastLED.show();
	leds[0] = CRGB(0, 0, 255); // blue
	FastLED.show();
	delay(200);
}
```
using `FastLED.addLeds<WS2812B, DATA_PIN,RGB>(leds, NUM_LEDS); // WS2812B pixel`
`leds[0] = CRGB(255,255,255); // White`
[image]

`leds[0] = CRGB(255, 0, 0);  // red/green depending on order`
[image]

`leds[0] = CRGB(0, 255, 0);  // red/green depending on order`
[image]

`leds[0] = CRGB(0, 0, 255); // blue`
[image]

[saleae logic 1.2.5 ](https://www.saleae.com/downloads)capture files : [capture.zip](https://github.com/FastLED/FastLED/files/189775/capture.zip)

I added the MKR1000 pin mapping to fastpin_arm_d21.h after the ARDUINO_SAMD_WINO's #elif
```
#elif defined(ARDUINO_SAMD_MKR1000)

#define MAX_PIN 22
_DEFPIN_ARM(  0, 0, 22); _DEFPIN_ARM(  1, 0, 23); _DEFPIN_ARM(  2, 0, 10); _DEFPIN_ARM(  3, 0, 11);
_DEFPIN_ARM(  4, 1, 10); _DEFPIN_ARM(  5, 1, 11); _DEFPIN_ARM(  6, 0, 20); _DEFPIN_ARM(  7, 0, 21);
_DEFPIN_ARM(  8, 0, 16); _DEFPIN_ARM(  9, 0, 17); _DEFPIN_ARM( 10, 0, 19); _DEFPIN_ARM( 11, 0,  8);
_DEFPIN_ARM( 12, 0,  9); _DEFPIN_ARM( 13, 1, 23); _DEFPIN_ARM( 14, 1, 22); _DEFPIN_ARM( 15, 0,  2);
_DEFPIN_ARM( 16, 1,  2); _DEFPIN_ARM( 17, 1,  3); _DEFPIN_ARM( 18, 0,  4); _DEFPIN_ARM( 19, 0,  5);
_DEFPIN_ARM( 20, 0,  6); _DEFPIN_ARM( 21, 0,  7);

#define SPI_DATA 8
#define SPI_CLOCK 9

#define HAS_HARDWARE_PIN_SUPPORT 1
```
as a comparison, this is what the WS2801 ( SPI ) output looks like - no missing byte
`leds[0] = CRGB(255,255,255); // White`
[image]

leds[0] = CRGB(255, 0, 0);  // red/green depending on order
[image]

leds[0] = CRGB(0, 255, 0);  // red/green depending on order
[image]

leds[0] = CRGB(0, 0, 255); // blue
[image]

```

focalintent, 2016-03-25:
What does it do for the second led - e.g. is it 0'ing out the first byte for all leds or just the first one - I don't have an m0 based setup here to test with right now, and won't be able to until later this weekend?

(#269 is a slightly different issue - or at least likely different cause - the code between avr and m0 arm is, to put it mildly, wildly different).

Daniel-dk, 2016-03-25:
The first byte of led[1] is also 0-ing out :  [image]](https://cloud.githubusercontent.com/assets/582384/14050112/5ae9f330-f2c2-11e5-8d1d-ade8a7684b8e.PNG)

focalintent, 2016-03-25:
Ah!  I see the problem w/the m0 code - i'll try checking in a fix shortly..

Daniel-dk, 2016-03-25:
Thank you ( We're building some MKR1000 based controllers- will send you one )

focalintent, 2016-03-25:
Try pulling master@HEAD and see if that fixes it.

Daniel-dk, 2016-03-25:
Works perfectly now, Thank you 
:
`leds[0] = CRGB(255,255,255);`
[image]
`leds[0] = CRGB(255,0,0); `
[image]

closed 2016-03-25 by focalintent