FastLED/FastLED #280: 2 instances on the same pin and isolation.

Vikaton, 2016-04-01:
2 questions:

1. Is it possible to isolate one LED strip from the other, so that something like `FastLED.setBrightness` only affects 1 of the 2 LED strips I'm using

2. is it possible to have 2 instances on the same pin so you would do `FastLED.addleds<NEOPIXEL, 6>(left, 24)` and another but with a different array of CRGB?

(The reason being is because I have a class that I want to use to control 2 LED strips at the same time (requires adding leds pin 6 and 7), and another class that just controls one( requires adding either pin or 7), I would want to be able to use both classes at the same time)

focalintent, 2016-04-01:
For the first one, you can do something like what's described here - https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples - basically bypassing the FastLED controller object to manage each strip separately.

```
void setup() { 
    FastLED.addLeds<WS2812, PIN1>(leds1, NUM_LEDS1);
    FastLED.addLeds<WS2812, PIN2>(leds2, NUM_LEDS2);
}

...

void loop() { 
...
    FastLED[0].show(strip1Brightness);
    FastLED[1].show(strip2Brightness);
```

As for the second one, no, you can't put two strips on the same pin.  I'd need to actually see your classes that you want to do to give you better advice on how to do what you want.

Vikaton, 2016-04-01:
Thank you for your response on the first question, here is my code for the 2nd one using Adafruit:

```cpp
#include <Adafruit_NeoPixel.h>

#define BRIGHTNESS 75

enum Color {
  red,
  green,
  blue,
  purple,
  pink,
  orange,
  white
};

class Both { // this is for using them both at the same time.
  public:
    Both(int lef, int righ);
    virtual void setColor(int col);
    void init();
    void setBright(int x);
    Adafruit_NeoPixel headL;
    Adafruit_NeoPixel headR;
  protected:
    uint32_t colors[7] = {headL.Color(255,0,0),
                    headL.Color(0,255,0),
                    headL.Color(0,0,255),
                    headL.Color(160,32,240),
                    headL.Color(255,192,203),
                    headL.Color(255,165,0),
                    headL.Color(255,255,255)};
    int color = red;
};

Both::Both(int lef, int righ) {
  headL = Adafruit_NeoPixel(24, lef, NEO_GRB + NEO_KHZ800);
  headR = Adafruit_NeoPixel(24, righ, NEO_GRB + NEO_KHZ800);
}
void Both::setColor(int col) {
  color = col;
  for (int i = 0; i<=24; i++) {
    headL.setPixelColor(i, colors[color]);
    headR.setPixelColor(i, colors[color]);
    headL.show();
    headR.show();
  }
}
void Both::init() {
  headL.begin();
  headL.show();
  headR.begin();
  headR.show();
}
void Both::setBright(int x) {
  headL.setBrightness(x);
  headR.setBrightness(x);
}

class Headlight : public Both {
  public:
    Adafruit_NeoPixel ring;
    Headlight(int pin);
    void showcase();
    virtual void setColor(int col);
    void fade();
    void heartbeat();
};

Headlight::Headlight(int pin) : Both(pin, 0) {
  ring = Adafruit_NeoPixel(24, pin, NEO_GRB + NEO_KHZ800);
}

void Headlight::setColor(int col) {
  color = col;
  for (int i = 0; i<=24; i++) {
    ring.setPixelColor(i, colors[color]);
    ring.show();
  }
}

Both ledring(6,7);  // left, right
Headlight left(6);

void setup() {
  left.ring.begin();
  left.ring.show();
  left.ring.setBrightness(BRIGHTNESS);
  ledring.init();
  ledring.setBright(BRIGHTNESS);
}

void loop() {
  left.setColor(red);
  delay(5000);
  ledring.setColor(blue);
  delay(5000);
}
```

As you can see, sometimes I use pin 7 for the parent class, and sometimes for the child one to isolate it without affecting the other LED ring.

focalintent, 2016-04-01:
So, one thing you can do with FastLED is:

```
#define NUM_LEDS_LEFT 24
#define NUM_LEDS_RIGHT 24

#define PIN_LEFT 6
#define PIN_RIGHT 7

CRGB leds[NUM_LEDS_LEFT + NUM_LEDS_RIGHT];
CRGB *leftLeds, *rightLeds;
void setup() { 
    leftLeds = leds;
    rightLeds = leds + NUM_LEDS_LEFT;
    FastLED.addLeds<NEOPIXEL, PIN_LEFT>(leftLeds, NUM_LEDS_LEFT);
    FastLED.addLeds<NEOPIXEL, PIN_RIGHT>(rightLeds, NUM_LEDS_RIGHT);
}

```

Now, when you want to make changes to all the leds, you can just work on the 48 element array, leds, and when you want to make changes to only one set, you can treat leftLeds or rightLeds as 24 element arrays.

In fact, this is how I do most of my pattern classes - I don't give them controllers, instead, I just give them leds, so for example:

```
class PatternBase { 
protected:
  CRGB *m_leds;
  uint16_t numLeds;
public:
  PatternBase(CRGB *leds, uint16_t numLeds) { setLeds(leds, numLeds); }

  void setLeds(CRGB *_leds, uint16_t _numLeds) { 
    m_leds = _leds;
    numLeds = _numLeds;
  }

  CRGB &operator[](uint16_t index) { return m_leds[index]; }

  virtual void draw() = 0;
};

class FadingDotPattern : public PatternBase {
  uint16_t x;
public:
  FadingDotPattern(CRGB *leds, uint16_t numLeds) : PatternBase(leds, numLeds) { x = 0; }

  virtual void draw() { 
    nscale8(leds, numLeds,192);
    if(x == numLeds) { x = 0; }
    leds[x] = CRGB::Red;
  }
};

CRGB leds[NUM_LEDS];
FadingDotPattern pattern(leds,NUM_LEDS);

void setup() { 
    FastLED.addLeds<NEOPIXEL,6>(leds, NUM_LEDS);
}

void loop() { 
  pattern.draw();
  FastLED.show();
  delay(20);
}
```

But then I could do something like, say:

```
  pattern.setLeds(leds, 10);
```

which resets the pattern object to only work on 10 leds, starting 5 leds in from the beginning of the leds array.  

Note that what i've done here is make it so that my pattern objects _only_ care about the CRGB objects they're working on - they don't care about controllers, or anything like that.  

Makes writing pattern code a helluva lot easier :)

Vikaton, 2016-04-01:
I think I got what you meat, but im not entirely sure, but I am closer to a solution!

here is what I have right now:

```cpp
#include "FastLED.h"

#define BRIGHTNESS 75

#define LEFT_LEDS 24
#define RIGHT_LEDS 24

#define PINR 7
#define PINL 6

class Both {
  public:
    Both(int lef, int righ);
    void setColor(CRGB lef, CRGB righ);
    void setColor(int col);
    void fade();
    CRGB all[RIGHT_LEDS + LEFT_LEDS];
    CRGB *headL, *headR;
};

Both::Both(int lef, int righ) {
  headL = all;
  headR = all + RIGHT_LEDS;
  FastLED.addLeds<NEOPIXEL, PINR>(headR, RIGHT_LEDS);
  FastLED.addLeds<NEOPIXEL, PINL>(headL, LEFT_LEDS);
}
void Both::setColor(CRGB lef, CRGB righ) {
  for (int i = 0; i<24; i++) {
    headL[i] = lef;
    headR[i] = righ;
  }
  FastLED.show();
}
void Both::setColor(int col) {
  for (int i = 0; i<48; i++) {
    all[i] = col;
  }
  FastLED.show();
}

class Headlight : public Both {
  public:
    Headlight(int pin);
    void setColor(CRGB col);
};

Headlight::Headlight(int pin) : Both(0,0) {
  ;
}

void Headlight::setColor(CRGB col) {
  for (int i = 0; i<24; i++) {
    headL[i] = col;
  }
  FastLED.show();
}

Both ledring(6,7);
Headlight left(7);  // 6: left, 7: right

void setup() {
  FastLED.setBrightness(BRIGHTNESS);
  ledring.setColor(CRGB::Purple, CRGB::Red);
  delay(3000);
  left.setColor(CRGB::White);
  FastLED.show();
}

void loop() {
}
```

The only problem is that which ever object instantiated last, thats the object will will work and light up (last night when I tried adding leds in both classes it just didnt light up at all)

So the problem is that everytime I instantiate the derived class after the parent one it "takes ownership" of the LEDs and the parent class cant do anything about it and vice versa, any pointers what what I have now?

focalintent, 2016-04-01:
The FastLED controllers work very differently from adafruit - you can't have multiple controllers on the same pin.  That's why I suggested having your classes _only_ work on arrays of CRGB objects.  

Alternatively, you could pass pointers to the individual led controllers (the one for pin 6, and the one for pin 7)

focalintent, 2016-04-01:
(Also - for the future, please take things like this to the g+ group - http://fastled.io/+ -- I try to keep the issues in github limited to bug reports and feature requests, not general programming help)

closed 2016-04-01 by focalintent

Vikaton, 2016-04-01:
Thank you, but I fixed it by declaring the `Both` class, then right after creating a global object of that class, THEN creating the other class for controlling a single LED that uses that global to avoid instantiating again, messy but it worked :))

Sorry for the misuse of issues of this repo, I was not aware.